2dom/PxMatrix #197: Expected maximum resolution with 1 Bit colour depth?

arahasya, 2020-05-16:
4. Running v 1.8.2 with esp32-Wroom32

I am using this library with monochrome displays and need Red colour only, what is the maximum display chain I can achieve using 1 bit colour depth?

I am using a 16x32 1/8 scan matrix.

Is there support for U-mapper for a display chain like below?
If I have a display structure like this :

1 -> 2 -> 3 -> 4
                         |
                        \/
8 <- 7 <- 6 <- 5

This was I can invert the displays from 5-8 and connect 4-5 instead of connecting a long cable from 4-8.

xsrf, 2020-05-16:
Hmm, the library is not really designed for monochrome displays. 1 bit is 1 bit per primary color and the framebuffer is still 16bit per pixel, so you're wasting lots of resources.
Your panel has only one data input, right? Have you already tried the library with one panel or is this hypothetical at the moment?

arahasya, 2020-05-16:
Well right now testing this library with rgb display manufactured by my company with connecting red colour only. 

[image]

Will it be very difficult to port to 1bit per pixel and 1 bit for framebuffer?

arahasya, 2020-05-16:
Current framebuffer is 16bit per pixel

So in theory if the framebuffer is reduced to 1bit per pixel can I achieve 16 times the resolution that is currently achieved?

xsrf, 2020-05-16:
Porting to 1 bit is not necessary for it to work. It is a huge waste of resources, but the ESP32 should handle that. You basically have 64x64 with 1/8 which is not a huge deal even with RGB.
Do you plan on using a bigger setup?

arahasya, 2020-05-16:
Yes I am running 16x352 1/8 on LPC2148 on 20Mhz
But that MCU is outdated now so I'm looking for another solution

xsrf, 2020-05-16:
Sorry btw, I have to correct myself, I mixed something up... The framebuffer is not 16bit per pixel. It uses 16 or 24bit color per pixel to draw but not all bits are saved in ram. When you use the library with 1 bit color depth, it saves only 1 bit per pixel per primary color.

closed 2020-06-12 by 2dom