2dom/PxMatrix #199: Display refresh rate on serial monitor

arahasya, 2020-05-17:
4. v1.8.2 Esp32

Is there any way to calculate and display refresh rate on the serial monitor? 

It is possible in this raspberry pi library 

https://github.com/hzeller/rpi-rgb-led-matrix/blob/daf8278946af1127ac8bc881a5ae037fb8fb4b86/lib/led-matrix.cc

xsrf, 2020-05-17:
A complete frame (one color bit only) is refreshed every time you call `display.display(x)`, so the refresh rate is dependent on how often you call `display.display(x)`. However, the runtime of `display.display(x)` is whats sets the limit here.
Lets do some ballpark calculations and assume a 16x352 1/8 Panel. 
On each call, data must be transferred for 704 pixels (one line), the LEDs have to be enabled for x microseconds, and this has to happen 8 times to complete the frame.
Lets assume your panel is okay with 20MHz SPI, then data transfer will take about (704x3)/20MHz = 106µs.
By default (x=30) for 1bit color, the LEDs are lit for 15µs.
So each call will take about 8x121µs = 968µs.
So basically you could achieve ~1000 refreshes per second.

If you use `setFastUpdate(true)` the LEDs are lit while data is transferred to optimize speed.
In this case the LEDs are lit at least as long as the transfer needs, so 106µs in the example above.
This will result in a brighter panel and a maximum refresh rate of ~1180Hz.

However, your code usually need some time to calculate what you are actually displaying, so refreshing every ~2-4ms should be reasonable.

As long as you call `display.display(x)` with a timer / ticker interrupt, the refresh rate will be always constant.
If you call it as part of your `loop()`, it will vary. However, this may cause visible flicker...

arahasya, 2020-05-17:
Thanks for this explanation

Why have you done (704x3)? Is that considering RGB?

xsrf, 2020-05-17:
Yes, this is because the library assumes and transfers RGB.

arahasya, 2020-05-17:
This is my attempt to remove green and blue computations. I have tested this with a single display

Works fine when I use panel width up till 4. Get no output when I exceed that. Why is the display getting limited at 16x128? Can you please check and tell me what I need to optimise more?

https://github.com/arahasya/PxMatrix

2dom, 2020-05-17:
Before we jump into that ... Could you post some close up pictures of your input and output connector?

2dom, 2020-05-17:
Would like to understand how these panels are structured first

arahasya, 2020-05-17:
So I had to use the Chinese 32x32 P6 1/8 panels because I have a lot of them

from esp32 MCU I connected A,B,C, OE, Latch, CLK and R1 only
output of the first display and all next displays with 16-pin FRC cable as 
R0 panel 1 -> R0 panel 2 -> R0 panel 3 ..............

[image]
[image]

Now I am getting output till 5 panels of 32x32 only but the program works till width 224 (chain=7)
but the data doesn't reach the 7th panel. Only till 5th

[image]

I have tried connecting to another display and use another wire to make sure.

I have also tried changing the height to 16 instead of 32 as I am using only the first 16 rows but then the text is mismatched because the matrix is 32x32 and still the text doesn't reach the 6th panel. But now the program works even with width (352 chain=11) but the text only reaches the 3rd panel

2dom, 2020-05-17:
Yeah...there is a default limitation of 64*64 pixels. So it makes sense that it does not work beyond panels

2dom, 2020-05-17:
You can specify the maximum matrix size by setting

#define PxMATRIX_MAX_HEIGHT 32
#define PxMATRIX_MAX_WIDTH 32*7

before you include PxMatrix.h

2dom, 2020-05-17:
As @xsrf mentioned the library is still treating this as an RGB panel  hence using 3x the resources it would have to. So we might introduce an option to drive monochrome displays so the lib only holds one colour in ram.

arahasya, 2020-05-17:
Yes I have already made those changes in width and height
Currently my visible text with red colour is 16x160 

I have tried to remove green and blue resources here
Not sure what else needs to be done

https://github.com/arahasya/PxMatrix/blob/master/test_mono.h

arahasya, 2020-05-17:
Do you know why the text mismatches when I use height as 16 instead of 32?

The data is anyway being sent only to the first 16 rows lower 16 rows aren't even connected.

arahasya, 2020-05-17:
Hey sorry my bad turns out the output connector was faulty of the last panel 

So here's another image with 7 panels workings now

[image]

So I am maxing out on 16x224

But I am wasting resources for lower half of 32x224

If I could find a way to make this work with height=16 or ignore the computational tasks for lower 16 rows I think I might achieve double the width I am doing right now

2dom, 2020-05-17:
Don't have a computer right now...changes would be reducing the buffer size in initialisation, disabling all code for green and blue in drawPixel, shifting out red component of buffer

arahasya, 2020-05-18:
I tried to do changes you suggested here. Right now just removed green and blue for testing instead of disabling. 

https://github.com/arahasya/PxMatrix/blob/master/test_mono.h

But still maybe I missed something

Output with 8 chains is this

[image]