2dom/PxMatrix #200: PxMatrix library question

Josiko85, 2020-05-17:
Good morning.
First thank you for your work.
Second, I wanted to ask you a question, I have an outdoor RBG panel with A and B pins only, and I scan 1/2, could you tell me if your library is valid for these panels? I can't get them to work.
Thank you very much.

rajeshdoshi, 2020-05-20:
I think you need to try 1/4

Josiko85, 2020-05-20:
ok , thank i tray to use it.

snow-leopard-climbing, 2020-06-05:
@Josiko85, hello. Please tell me if you could make the panel work only with A, B pins using Arduino. I have the same problem and cannot find a solution for 2 months ...

2dom, 2020-06-05:
What is the problem?

snow-leopard-climbing, 2020-06-05:
I have a rgb panel with the inputs in the photo. 
Panel: P10 Outdoor 32×16 SMD Scan 1/2 Qiangli Q10C2V5H.
SPECIFICATION: http://pmod86499.pic35.websiteonline.cn/upload/vdy5.pdf

The problem is that all libraries use  a, b, c pins. But I have a, b pins only.

[image]

2dom, 2020-06-06:
Oh...you can also use the library with just A and B Pins. Simply remove the C pin number from the constructor.

2dom, 2020-06-06:
inline PxMATRIX(uint16_t width, uint16_t height,uint8_t LATCH, uint8_t OE, uint8_t A,uint8_t B);

2dom, 2020-06-06:
Or just use the original constructor and do not connect C

Josiko85, 2020-06-06:
Yes, that's the problem, a p10 panel without C pin, and with scand 1/2, but I also have another one, I use arduino mega, and in all the example codes I have:

#ifdef ESP32 or #ifdef ESP8266

so I don't get to compile it for arduino mega,

If I've seen the connections, thank you, but could you give an example using mega?

Thank you for your time and help.

2dom, 2020-06-06:
There is an atmel example in the example programs - did you check that one out?

Josiko85, 2020-06-06:
the example is pixeltime_AVR? that example is the only one I've been able to compile with mega
if this is it, this weekend I will try to test it by doing the installation with the panel

2dom, 2020-06-06:
Yes...that's for Atmel. It does not use interrupts to call the display update so it is slightly different to the ESP32 and esp8266 examples

Josiko85, 2020-06-06:
Okay, thank you very much. I'll try to get it working and let you know about the progress and problems. Thank you

snow-leopard-climbing, 2020-06-06:
@2dom, thanks for your response!

  I use Arduino Uno and RGB panel (P10 32×16  Scan 1/2), which has only A, B pins.
  I connected RGB panel as written in the Redmi file. Then I compiled the pixeltime_AVR example with the following settings:
`...`

`// Pins for LED MATRIX`
`#define P_A 2`
`#define P_B 3`
`#define P_C 0`
`#define P_D 0`
`#define P_E 0`
`#define P_LAT 7`
`#define P_OE 8`

`PxMATRIX display(32, 16, P_LAT, P_OE, P_A, P_B);`

`Serial.begin(9600);`
`display.begin(2);`
`display.setMuxPattern(STRAIGHT);`
`display.setScanPattern(LINE);`

 `...`

But the results are incorrect. The text “Pixel Time” is initially displayed on the panel (in the photo below).

[image]

I tried different combinations of settings (

`begin{2, 4, 8}, `
`setMuxPattern{BINARY, STRAIGHT}, `
`setScanPattern{LINE, ZIGZAG,ZZAGG, ZAGGIZ, WZAGZIG, VZAG, ZAGZIG}`

), but the result does not improve...

Josiko85, 2020-06-06:
what pin  do you use for R1 in arduino uno?

snow-leopard-climbing, 2020-06-06:
My RGB panel input connector has "R0" in the top left corner. I connected it to the 11th pin on Arduino Uno.

Josiko85, 2020-06-06:
Right, you're right, I didn't get a good look at him

Josiko85, 2020-06-06:
i have this problem.
[https://youtu.be/C3sSRUXberI ](url)

2dom, 2020-06-06:
Please try it with

```
display.begin{4}
display.setMuxPattern{BINARY}
```

and try all the different scan patterns

Josiko85, 2020-06-06:
i use
#define PxMATRIX_COLOR_DEPTH 1
//#define PxMATRIX_MAX_HEIGHT 32
//#define PxMATRIX_MAX_WIDTH 64
#define PxMATRIX_MAX_WIDTH 32
#define PxMATRIX_MAX_HEIGHT 16
  and with display.begin(4); the result is the same but only in rows 5-8 and 13-16 

 and with display.begin(8) I only have half of the left panel but all the rows

and from 12 just like in the video

can i put a code to turn on a led only?

I'll continue the tests tomorrow, thanks.

snow-leopard-climbing, 2020-06-06:
@2dom, 
 I tried as you requested:
`display.begin{4};`
`display.setMuxPattern{BINARY};`
and all the different scan patterns. Everywhere result looks like this:
[image]

I also tried with `display.begin{2};` Everywhere result looks like this:
[image]

2dom, 2020-06-06:
OK ... lets take a step back. Could you call 

display.displayTestPattern(20);

instead of display.display and post the results (video or gif)

snow-leopard-climbing, 2020-06-06:
@2dom, 

OK, I'm going to try now: display.displayTestPattern(20);

And
I decided to run with the following settings:
`...`

`display.begin(2);`
`display.setMuxPattern(BINARY);`
`display.setScanPattern(ZAGZIG);`

`...`

`  display.clearDisplay();`
`  display.setTextColor(myCYAN);`
`  display.setCursor(2,0);`
`  display.print("1");`

`...`

 and got an interesting result:
[image]

The number 1 seems to be divided into 2 halves...

snow-leopard-climbing, 2020-06-06:
@2dom, 
I recorded a video when `display.displayTestPattern(20)` instead of `display.display(20)`
https://youtu.be/pJadKCt94r8

2dom, 2020-06-06:
This is an rather unusual panel 

Can you try the pixel time example again with

display.begin(2);
display.setMuxPattern(STRAIGHT);

and ZAGZIG or ZIGZAG

2dom, 2020-06-06:
OK ... just had a think about this pattern. It is not supported right now. Will try to implement when I have the time.

Josiko85, 2020-06-06:
> @2dom,
>
> OK, I'm going to try now: display.displayTestPattern(20);
>
> And
> I decided to run with the following settings:
> `...`
>
> `display.begin(2);`
> `display.setMuxPattern(BINARY);` [...]

Could you putt the complete code? 
please

snow-leopard-climbing, 2020-06-07:
@2dom, , thanks for your response!

The code below gives an interesting result (in the photo). Figures 1 and 7 are cut into two halves ...

`#define PxMATRIX_COLOR_DEPTH 1`
`#define PxMATRIX_MAX_HEIGHT 16`
`#define PxMATRIX_MAX_WIDTH 32`

`#include <PxMatrix.h>`
`#include <avr/pgmspace.h>`
`// Pins for LED MATRIX`
`#define P_A 2`
`#define P_B 3`
`#define P_C 0`
`#define P_D 0`
`#define P_E 0`
`#define P_LAT 7`
`#define P_OE 8`

`PxMATRIX display(32, 16, P_LAT, P_OE, P_A, P_B);`

`// Some standard colors`
`uint16_t myCYAN = display.color565(0, 255, 255);`
`uint16_t myMAGENTA = display.color565(255, 0, 255);`

`unsigned long start_time=0;`

`void setup() {`

`  Serial.begin(9600);`

`  display.begin(2);`
`  display.setMuxPattern(BINARY);`
`  display.setScanPattern(ZAGZIG);`

`  display.clearDisplay();`
`  display.setTextColor(myCYAN);`
`  display.setCursor(2,0);`
`  display.print("1");`

`  display.setTextColor(myMAGENTA);`
`  display.setCursor(2,8);`
`  display.print("7");`

`  start_time = millis();`
`  while((millis()-start_time)<300000000)`
`  display.display(20);`
`}`

`void loop() `
`{`
`}`

[image]

snow-leopard-climbing, 2020-06-07:
@2dom, 
I did some experiments, I hope this helps to understand the problem.

1) I ran the same code as in the comment above. But instead of `display.setMuxPattern(BINARY);`
 I wrote `display.setMuxPattern(STRAIGHT);`. Double pixels disappeared (result in ph
[image]
oto)

snow-leopard-climbing, 2020-06-07:
2)I ran the following code for different ScanPattern{LINE, ZIGZAG, ZAGGIZ, ZAGZIG}

    #define PxMATRIX_COLOR_DEPTH 1
    #define PxMATRIX_MAX_HEIGHT 16
    #define PxMATRIX_MAX_WIDTH 32

    #include <PxMatrix.h>
    #include <avr/pgmspace.h>

    // Pins for LED MATRIX
    #define P_A 2
    #define P_B 3
    #define P_C 0
    #define P_D 0
    #define P_E 0
    #define P_LAT 7
    #define P_OE 8

    PxMATRIX display(32, 16, P_LAT, P_OE, P_A, P_B);

    // Some standard colors
    uint16_t myCYAN = display.color565(0, 255, 255);
    uint16_t myMAGENTA = display.color565(255, 0, 255);

    unsigned long start_time=0;

    void setup() 
    {
        Serial.begin(9600);
        display.begin(2);
        display.setMuxPattern(STRAIGHT);
        display.setScanPattern(ZAGZIG);

        display.clearDisplay();

        for (int y=0; y<16; y++)
        {
            for (int x=0; x<32; x++)
            {
              display.drawPixel(x, y, myCYAN);

              start_time = millis(); 
              while((millis()-start_time)<80)
                  display.display(20);
            }
        }
    }

    void loop()
    {
    }

Results
LINE: https://youtu.be/zMmyp76_AJ0
ZIGZAG: https://youtu.be/RmbcMRaYNlo
ZAGGIZ: https://youtu.be/dnMBt4EbJnE
ZAGZIG: https://youtu.be/pAlmSIoZTtY

Josiko85, 2020-06-07:
> @2dom,
> 2) I ran the following code for different ScanPattern{LINE, ZIGZAG, ZAGGIZ, ZAGZIG}
> 
> ```
> #define PxMATRIX_COLOR_DEPTH 1
> #define PxMATRIX_MAX_HEIGHT 16
> #define PxMATRIX_MAX_WIDTH 32
> 
> #include <PxMatrix.h>
> #include <avr/pgmspace.h>
> 
> // Pins for LED MATRIX
> #define P_A 2
> #define P_B 3
> #define P_C 0
> #define P_D 0
> #define P_E 0
> #define P_LAT 7
> #define P_OE 8
> 
> PxMATRIX display(32, 16, P_LAT, P_OE, P_A, P_B);
> 
> // Some standard colors
> uint16_t myCYAN = display.color565(0, 255, 255);
> uint16_t myMAGENTA = display.color565(255, 0, 255);
> 
> unsigned long start_time=0;
> 
> void setup() 
> {
>     Serial.begin(9600);
>     display.begin(2);
>     display.setMuxPattern(STRAIGHT);
>     display.setScanPattern(ZAGZIG);
> 
>     display.clearDisplay();
> 
>     for (int y=0; y<16; y++)
>     {
>         for (int x=0; x<32; x++)
>         {
>           display.drawPixel(x, y, myCYAN);
> 
>           start_time = millis(); 
>           while((millis()-start_time)<80)
>               display.display(20);
>         }
>     }
> }
> 
> void loop()
> {
> }
> ```
> 
> Results
> LINE: https://youtu.be/zMmyp76_AJ0
> ZIGZAG: https://youtu.be/RmbcMRaYNlo
> ZAGGIZ: https://youtu.be/dnMBt4EbJnE
> ZAGZIG: https://youtu.be/pAlmSIoZTtY

What is your panel model? Because I have a problem with the odd line.

My model is p10-3535-2s

[https://www.youtube.com/watch?v=9qTppYEpw5g](url)

Josiko85, 2020-06-07:
but if i use .
 display.setMuxPattern(BINARY);
It works correctly

snow-leopard-climbing, 2020-06-07:
@2dom, 
having carefully watched the video (runing code as in my comment above with ScanPattern = ZAGZIG: https://youtu.be/pAlmSIoZTtY), I understood the sequence.
For some reason, the display is divided into 8 parts and if you swap 1 and 4, as well as 5 and 8, then the panel will show correctly.

The blue color at the top shows: "1 2".
The violet color at the bottom shows: "7 8".

[image]

    #define PxMATRIX_COLOR_DEPTH 1
    #define PxMATRIX_MAX_HEIGHT 16
    #define PxMATRIX_MAX_WIDTH 32

    #include <PxMatrix.h>
    #include <avr/pgmspace.h>

    // Pins for LED MATRIX
    #define P_A 2
    #define P_B 3
    #define P_C 0
    #define P_D 0
    #define P_E 0
    #define P_LAT 7
    #define P_OE 8

    PxMATRIX display(32, 16, P_LAT, P_OE, P_A, P_B);

    // Some standard colors
    uint16_t myCYAN = display.color565(0, 255, 255);
    uint16_t myMAGENTA = display.color565(255, 0, 255);

    unsigned long start_time=0;
    uint8_t icon_index=0;

    void setup() {

         Serial.begin(9600);

          display.begin(2);
          display.setMuxPattern(STRAIGHT);
          display.setScanPattern(ZAGZIG);

          display.clearDisplay();
  
          display.setTextColor(myCYAN);
          display.setCursor(2,0);
          display.print("1  2");

          display.setTextColor(myMAGENTA);
          display.setCursor(2,8);
          display.print("7  8");
 
          start_time = millis();
          while((millis()-start_time)<300000000)
              display.display/*TestPattern*/(20);

      }

      void loop() 
      {
      }

snow-leopard-climbing, 2020-06-07:
@Josiko85, my panel: P10 Outdoor 32×16 SMD Scan 1/2 Qiangli Q10C2V5H.
SPECIFICATION: http://pmod86499.pic35.websiteonline.cn/upload/vdy5.pdf

If I turn on `display.setMuxPattern(BINARY);`, then I have double pixels.

Josiko85, 2020-06-07:
> @2dom,
> having carefully watched the video (runing code as in my comment above with ScanPattern = ZAGZIG: https://youtu.be/pAlmSIoZTtY), I understood the sequence.
> For some reason, the display is divided into 8 parts and if you swap 1 and 4, as well as 5 and 8, then the panel will show correctly.
> 
> The blue color at the top shows: "1 2".
> The violet color at the bottom shows: "7 8".
> 
> [image]
> 
> ```
> #define PxMATRIX_COLOR_DEPTH 1
> #define PxMATRIX_MAX_HEIGHT 16
> #define PxMATRIX_MAX_WIDTH 32
> 
> #include <PxMatrix.h>
> #include <avr/pgmspace.h>
> 
> // Pins for LED MATRIX
> #define P_A 2
> #define P_B 3
> #define P_C 0
> #define P_D 0
> #define P_E 0
> #define P_LAT 7
> #define P_OE 8
> 
> PxMATRIX display(32, 16, P_LAT, P_OE, P_A, P_B);
> 
> // Some standard colors
> uint16_t myCYAN = display.color565(0, 255, 255);
> uint16_t myMAGENTA = display.color565(255, 0, 255);
> 
> unsigned long start_time=0;
> uint8_t icon_index=0;
> 
> void setup() {
> 
>      Serial.begin(9600);
> 
>       display.begin(2);
>       display.setMuxPattern(STRAIGHT);
>       display.setScanPattern(ZAGZIG);
> 
>       display.clearDisplay();
> 
>       display.setTextColor(myCYAN);
>       display.setCursor(2,0);
>       display.print("1  2");
> 
>       display.setTextColor(myMAGENTA);
>       display.setCursor(2,8);
>       display.print("7  8");
> 
>       start_time = millis();
>       while((millis()-start_time)<300000000)
>           display.display(20);
> 
>   }
> 
>   void loop() 
>   {
>   }
> ```

i can see a different kind of panel becouse if i wrote your code i can saw the number but in different place, but i can see all the part of number

Josiko85, 2020-06-07:
i have good news, using the  next code ` 
      x=1;
      y=0;
      display.drawPixel(x, y, myMAGENTA);
          
          start_time = millis(); 
          while((millis()-start_time)<1000)
          display.display(5);
          display.clearDisplay();`

I can draw a pixel if I repeat with another one I can draw a good drawing, now I am sequencing all the pixels of my panel, at least so I can draw.

Josiko85, 2020-06-08:
> What is your panel model? Because I have a problem with the odd line.
>
> My model is p10-3535-2s
>
> [https://www.youtube.com/watch?v=9qTppYEpw5g](url)

thes is my matrix secuence [https://studio.youtube.com/video/0P-b3NP8EKM/edit](url)

it is different that your

2dom, 2020-06-11:
Well zaggiz is looking good. Do you want to try the Pixel Time example again with these settings?

snow-leopard-climbing, 2020-06-11:
@2dom, thank you for your time!
I I downloaded the last master and started with

     display.begin(2);
     display.setMuxPattern(STRAIGHT);
     display.setScanPattern(ZAGZIG);
     display.setBlockPattern(DBCA);.

The result is a little different. I also display "1   2" at the top and "7   8" at the bottom of the panel. Some parts are reversed (if you compare the result without `display.setBlockPattern(DBCA);`)

The result with `display.setBlockPattern(DBCA);`in the photo:
[image]

snow-leopard-climbing, 2020-06-11:
@2dom, and the result of the script for the white lines looks the same as I previously shot a video without `display.setBlockPattern(DBCA);`: https://youtu.be/pAlmSIoZTtY

2dom, 2020-06-11:
@snow-leopard-climbing ... typo ... can you try again with the latest master?

snow-leopard-climbing, 2020-06-11:
@2dom,
Hooray!!!! Thank you very much! You have solved a big problem for the display with only A, B pins!

pixeltime_AVR work correctly!

And result my script: https://youtu.be/Ehfx5z3i8yc

2dom, 2020-06-11:
No worries man...glad that it works now :)

Josiko85, 2020-06-11:
@2dom You are the best
https://youtu.be/Cg2wost5eYY

the code is:#define PxMATRIX_COLOR_DEPTH 1
#define PxMATRIX_MAX_HEIGHT 16
#define PxMATRIX_MAX_WIDTH 32

#include <PxMatrix.h>
#include <avr/pgmspace.h>

// Pins for LED MATRIX
#define P_A 2
#define P_B 3
#define P_C 0
#define P_D 0
#define P_E 0
#define P_LAT 7
#define P_OE 8

PxMATRIX display(32, 16, P_LAT, P_OE, P_A, P_B);

// Some standard colors
uint16_t myRED = display.color565(255, 0, 0);
uint16_t myGREEN = display.color565(0, 255, 0);
uint16_t myBLUE = display.color565(0, 0, 255);
uint16_t myWHITE = display.color565(255, 255, 255);
uint16_t myYELLOW = display.color565(255, 255, 0);
uint16_t myCYAN = display.color565(0, 255, 255);
uint16_t myMAGENTA = display.color565(255, 0, 255);
uint16_t myBLACK = display.color565(0, 0, 0);

unsigned long start_time=0;

void setup() 
{
    Serial.begin(9600);
    display.begin(2);
    display.setMuxPattern(BINARY);
//  display.setScanPattern(ZIGZAG);
    display.setScanPattern(ZAGGIZ);
//  display.setScanPattern(ZAGZIG);
    display.setBlockPattern(DBCA);
    
//    display.clearDisplay();
//
//    for (int y=0; y<16; y++)
//    {
//        for (int x=0; x<32; x++)
//        {
//          display.drawPixel(x, y, myCYAN);
//
//          start_time = millis(); 
//          while((millis()-start_time)<80)
//              display.display(20);
//        }
//    }

 display.clearDisplay();
  display.setTextColor(myCYAN);
  display.setCursor(2,0);
  display.print("Pixel");
  display.setTextColor(myMAGENTA);
  display.setCursor(2,8);
  display.print("Time");
  start_time = millis();
  while((millis()-start_time)<3000)
    display.display(20);
}

void scroll_text()
{
    display.setTextWrap(false);  // we don't wrap text so it scrolls nicely
    display.setTextSize(2);
    display.setRotation(0);

    for (int xpos=64; xpos>-300; xpos--)
    {
      display.setTextColor(myWHITE);
      display.clearDisplay();
      display.setCursor(xpos,0);
      display.println("Welcome to PxMatrix!   ");
      start_time = millis();
      while((millis()-start_time)<100)
           display.display(5);
    }
}

void loop()
{ display.clearDisplay();
  scroll_text();
}
``

2dom, 2020-06-11:
@Josiko85 Great that it works now...could you remove the display.setBlockPattern() command and try one more time. I think with your display the blocking may not be needed...

Josiko85, 2020-06-11:
> 
> 
> @Josiko85 Great that it works now...could you remove the display.setBlockPattern() command and try one more time. I think with your display the blocking may not be needed...

You're right, it doesn't have to work perfectly, I'll find out if it's possible to send you a virtual coffee

Josiko85, 2020-06-11:
You have a donation from me. 
Thank you very much for everything.

2dom, 2020-06-11:
You could buy me a beer - there is a beer do donation button on the GitHub page. Enjoy your Matrix!

Josiko85, 2020-06-11:
> 
> 
> You could buy me a beer - there is a beer donation button on the GitHub page. Enjoy your Matrix!

inform me if you don't get my donation I used Paypal

snow-leopard-climbing, 2020-06-11:
@2dom 
may I ask another question? I connected two displays as in the picture below and it is written in Readme.txt.
It works as one large display, but for some reason the old problem has remained ...

[image]

    #define PxMATRIX_COLOR_DEPTH 1
    #define PxMATRIX_MAX_HEIGHT 16
    #define PxMATRIX_MAX_WIDTH 64

    #include <PxMatrix.h>
    #include <avr/pgmspace.h>
    // Pins for LED MATRIX
    #define P_A 2
    #define P_B 3
    #define P_C 0
    #define P_D 0
    #define P_E 0
    #define P_LAT 7
    #define P_OE 8

    PxMATRIX display(64, 16, P_LAT, P_OE, P_A, P_B);

    // Some standard colors
    uint16_t myCYAN = display.color565(0, 255, 255);

    unsigned long start_time=0;
    uint8_t icon_index=0;

    void setup() {

        Serial.begin(9600);
        display.begin(2);
        display.setMuxPattern(STRAIGHT);
        display.setScanPattern(ZAGZIG);
        display.setBlockPattern(DBCA);
  
        display.clearDisplay();
  
        for (int y=0; y<16; y++)
        {
            for (int x=0; x<64; x++)
            {
                display.drawPixel(x, y, myCYAN);
      
                start_time = millis();
      
                while((millis()-start_time)<80)
                    display.display(20);   
            }
        }
    }

    void loop() {

    }

snow-leopard-climbing, 2020-06-11:
@2dom, can take a look at the video, please: https://youtu.be/K1QvLEZD2FA

2dom, 2020-06-11:
You will have to set 

`display.setPanelsWidth(2); `

to chain two panels.

2dom, 2020-06-11:
@Josiko85 Thanks for the beer! It arrived safely in Germany - cheers!

Josiko85, 2020-06-11:
@2dom with more panels doesn't work well, can you look at this example?
https://youtu.be/bnkWSbs95sc

2dom, 2020-06-11:
Can you run the line test example one more time?

```
 for (int y=0; y<16; y++)
    {
        for (int x=0; x<64; x++)
        {
            display.drawPixel(x, y, myCYAN);
  
            start_time = millis();
  
            while((millis()-start_time)<80)
                display.display(20);   
        }
    }
```

snow-leopard-climbing, 2020-06-11:
@2dom, yes, I forgot the `display.setPanelsWidth(2);`. But then I added it.
I display the numbers "1 2 3 4" at the top of the panel and "5 6 7 8" at the bottom of the panel.

    #define PxMATRIX_COLOR_DEPTH 1
    #define PxMATRIX_MAX_HEIGHT 16
    #define PxMATRIX_MAX_WIDTH 64

    #include <PxMatrix.h>
    #include <avr/pgmspace.h>
    // Pins for LED MATRIX
    #define P_A 2
    #define P_B 3
    #define P_C 0
    #define P_D 0
    #define P_E 0
    #define P_LAT 7
    #define P_OE 8

    PxMATRIX display(64, 16, P_LAT, P_OE, P_A, P_B);

    // Some standard colors
    uint16_t myCYAN = display.color565(0, 255, 255);
    uint16_t myMAGENTA = display.color565(255, 0, 255);

    unsigned long start_time=0;
    uint8_t icon_index=0;

    void setup() {

         Serial.begin(9600);

         display.begin(2);
         display.setMuxPattern(STRAIGHT);
         display.setScanPattern(ZAGZIG);
         display.setBlockPattern(DBCA);
         display.setPanelsWidth(2);

         display.clearDisplay();

         display.setTextColor(myCYAN);
         display.setCursor(2,0);
         display.print("1  2  3  4");

         display.setTextColor(myMAGENTA);
         display.setCursor(2,8);
         display.print("5  6  7  8");
 
         start_time = millis();
         while((millis()-start_time)<300000000)
             display.display(20);
    }

    void loop() { }

Here are the results of some experiments:
1) The code above is unchanged.

[image]

2) Code above without `display.setPanelsWidth(2);`

[image]

3) Code above without `display.setBlockPattern(DBCA);`

[image]

Josiko85, 2020-06-11:
> Can you run the line test example one more time?
>
> ```
>  for (int y=0; y<16; y++)
>     {
>         for (int x=0; x<64; x++)
>         {
>             display.drawPixel(x, y, myCYAN); [...]
this is the video
https://youtu.be/LLLfRnfLU68

xsrf, 2020-06-11:
> @2dom with more panels doesn't work well, can you look at this example?
> https://youtu.be/bnkWSbs95sc

You've initialized your panel as 32x16. You must set it to 64x16.

Josiko85, 2020-06-11:
Yes, I've tried but the result is the same and the pattern is the same

> 
> 
> > @2dom with more panels doesn't work well, can you look at this example?
> > https://youtu.be/bnkWSbs95sc
> 
> You've initialized your panel as 32x16. You must set it to 64x16.

Yes, I've tried but the result is the same and the pattern is the same

2dom, 2020-06-11:
Can you guys tell me how you chained the panels? You simply use the supplied flat band cable connector between panels ...

[28 more replies]

closed 2020-06-12 by 2dom