2dom/PxMatrix #201: 64x64 please help to setup properly

Nikitok122, 2020-05-18:
hello everyone, having that module p3.91 

https://www.aliexpress.com/item/33013222591.html?spm=a2g0o.productlist.0.0.309d42f4694THF&algo_pvid=de08efee-aebd-40d4-93e7-03852b40c5b9&algo_expid=de08efee-aebd-40d4-93e7-03852b40c5b9-3&btsid=0ab50f4415897715389322401e6874&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

tried many different settings but still the same.  Main problem is doubled display as showed on video

https://www.youtube.com/watch?v=TYyFVypr3sA

arahasya, 2020-05-18:
You probably haven't specified the correct height and width or chain

Post your code

Nikitok122, 2020-05-18:
I used code from examples, post it later. But I did different changes and about matrix resolution and chaining too. But wanna give one clarification about my panel: it has resolution 64x64 but without E line, only D. Maybe it has some chaining inside the display

Nikitok122, 2020-05-18:
And changes I did in code to setup display didn't affect to that doubling

xsrf, 2020-05-18:
I'd recommend to try with only one panel first until you know your settings are right.
4 address lines suggest it's a 1/16 panel.

Nikitok122, 2020-05-18:
This is one panel 64x64. Yes, in settings it was selected as 1/16

xsrf, 2020-05-18:
oh, right. looked like four ;) Can you try pattern_test.ino and post a video please? Also, the pictures on ali only show one connector. Can you show your wiring?

Nikitok122, 2020-05-18:
Pinot of that panel according that pic
https://hackster.imgix.net/uploads/image/file/146124/DisplayPinout.jpg?auto=compress%2Cformat&w=1280&h=960&fit=max

I'll post that video tonight

Nikitok122, 2020-05-18:
`#include <PxMatrix.h>

#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#endif
// Pins for LED MATRIX

//PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);
PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D);
//PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

#ifdef ESP8266
// ISR for display refresh
void display_updater()
{
  display.displayTestPattern(70);
  // display.displayTestPixel(70);
  //display.display(70);
}
#endif

#ifdef ESP32
void IRAM_ATTR display_updater(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  //isplay.display(70);
  display.displayTestPattern(70);
  portEXIT_CRITICAL_ISR(&timerMux);
}
#endif

uint16_t myCYAN = display.color565(0, 255, 255);
void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
  display.begin(16);
   display.setDriverChip(SHIFT);
  display.setColorOrder(RRGGBB);
  display.flushDisplay();
  display.setTextColor(myCYAN);
  display.setCursor(2,0);
  display.print("Pixel");
  Serial.println("hello");

  #ifdef ESP8266
    display_ticker.attach(0.004, display_updater);
  #endif

  #ifdef ESP32
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &display_updater, true);
    timerAlarmWrite(timer, 4000, true);
    timerAlarmEnable(timer);
  #endif

  delay(1000);
}

void loop() {

 delay(100);

}`

Nikitok122, 2020-05-18:
https://www.youtube.com/watch?v=X5fVU3vlAmA video of pattern test

xsrf, 2020-05-19:
Can you run the test for longer? 2-3min? Also, are the LEDs lit in different colors? It's hard to see because of the over exposure. Can you give more details? You can reduce the brightness with `displayTestPattern(10);` or may also cover it with paper for filming.

Nikitok122, 2020-05-20:
[image]

Nikitok122, 2020-05-20:
yes, different color. yes, that test could be run as long as u want it

2dom, 2020-06-12:
Oh ... that does not look good. It appears that the colors change with every bit. Is is completely different to how other panels are working. Not supported at the moment I am afraid

closed 2020-06-12 by 2dom