2dom/PxMatrix #203: about multi P6 RGB panel

adincergok, 2020-05-19:
Hello everybody,

I wanna use 6 pieces P6 Panel with ESP8266 (nodemcu).
1 panel work good, 2 panel work good but;
when I connect to 3th panel, it not work. Some pixels fails, some shadows etc.
so when I connected 6th panel, no any image.

ESP8266 pinout is 3.3V and 12mA, i think its not enough for last panel. and I think, I have to use any module for missing voltage. 

Note: if I shorten cables, image getting better. but i never connect 4 th panel.

any suggestion?

thanks.

xsrf, 2020-05-20:
The digital signal from the ESP is buffered on the panels themselves. All digital outputs of the first panel should be 5V (or whatever Vcc you use) and only need to drive the next input. So the amount of panels should not change anything regarding the driving capacity of the ESP. The weak link is to the first panel only. Can you show your setup/cabling and maybe videos of 1 panel and 2 or 3 to compare with? Also, if you can, lower the Vcc of your panels to around 4.5V instead of 5V. This may help.

2dom, 2020-05-20:
Connecting a large number of panels also leads to pixel shifts due to signal delays within the panels. https://twitter.com/MLE_Online/status/1192264866158899200?s=19 ... You may want to consider another library if you need this

rajeshdoshi, 2020-05-20:
Practically what is the maximum number of panels one can use?

xsrf, 2020-05-20:
@rajeshdoshi it depends on the panel hardware

It is correct that colors may shift pixel by pixel with a lot of panels. This is because the Clock-Signal for the first panel comes from the microcontroller whereas the data-signals for R2,G1,G2,B1,B2 have to get through all panels.
Testing for this issue is easy. First of all, the R1 channel will never be affected by this issue regardless how many panels you have. The R2 channel will be least affected, the B2 channel will be affected most. Also, the pixel shift would be konsistent and it will only happen within the same row of the panel.
If you suspect this is an issue, lowering the SPI data rate will solve it (but it will reduce the max refresh rate possible). Use `#define PxMATRIX_SPI_FREQUENCY 10e6` for 10MHz instead of 20MHz.

A more permanent solution for this is using a library that drives R1/R2/G1/G2/B1/B2 in parallel which is not possible on the ESP8266 due to lack of IO OR by changing the wiring to loop all data entirely through all channels of the first display before providing it to the next panel. However, this would require a different layout for the framebuffer in the library.

xsrf, 2020-05-20:
Also, because all signals are refreshed after each panel, the error would be really consistent. The shift will only happen on the loop-back from one color channel output to the next input and it should be the same for all channels. Meaning, if R2 is shifted by one, it should be shifted by one pixel for all panels. And if R2 is shifted by one pixel, all following channels will also be shifted by one pixel with respect to the previous color. Only if the delay matches the clock frequency, it may kind of jitter.

xsrf, 2020-05-20:
That's what may happen here in the Video @2dom mentioned:
https://twitter.com/MLE_Online/status/1192265470117695488
These panels are rotated 90°, a row goes from top to bottom. You can see that the shift is kind of consistent one pixel per color channel.

adincergok, 2020-05-20:
> Practically what is the maximum number of panels one can use?

I use 6 panels.

adincergok, 2020-05-20:
> The digital signal from the ESP is buffered on the panels themselves. All digital outputs of the first panel should be 5V (or whatever Vcc you use) and only need to drive the next input. So the [...]

xsrf, thank you. I understand you well. first panel drive via ESP output (3.3V) than other panel drive via this panel. so 2, 3, ... panels are always drive with 5V. I have to correctly connect cables.

I checked repeatedly connection. but this night (in 8-9 hours) I put photos and videos about connections, panels and power supply.

adincergok, 2020-05-20:
> Connecting a large number of panels also leads to pixel shifts due to signal delays within the panels. https://twitter.com/MLE_Online/status/1192264866158899200?s=19 ... You may want to consider another library if you need this

What do you advice me?

adincergok, 2020-05-20:
> @rajeshdoshi it depends on the panel hardware
>
> It is correct that colors may shift pixel by pixel with a lot of panels. This is because the Clock-Signal for the first panel comes from the [...]

i will try to lowering data rate this night. i will put video. my power suplly 5V i may lower this to 4.5V. i will try all your advices.

adincergok, 2020-05-21:
yesterday I work about it all night. but many problem. I write here my problems cause i need your ideas. And I will put videos and images how I connect, what I use.

1. My P6 panel hub is not HUB75.  it is HUB75D (is it important!! i dont know)
[image]

2. Last panel always take parasitic from my hand (when 6 panel connected)

3. @xsrf, I use power supply 5V and 55A. and I reduce voltage to 4.5V
[image]
[image]

4. I connect Input and Output pins like PxMatrix main page shown.
input side
[image]

output side
[image]

and connect output side 5 pins to input side
[image]

5. I was trying to connect output side GND. it didn't work.

6. ESP8266 (NodeMCU) connections
[image]

7. Also I try to use external power supply (5V in Vin). but its insignificant. it didnt work. USB is enough for ESP MCU
[image]

8. 1 piece panel working
   8.a.  my panel back ground
[image]

my Code:  (i cleared remarks)

1 PANEL
==========================================================
#define PxMATRIX_double_buffer true
#include <PxMatrix.h>

// Pins for LED MATRIX
#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#define matrix_width 32
#define matrix_height 16

uint8_t display_draw_time=60; //30-70 is usually fine

//PxMATRIX display(96,32,P_LAT, P_OE,P_A,P_B,P_C);
PxMATRIX display(32,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
//PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

// Some standard colors
uint16_t myRED = display.color565(255, 0, 0);
uint16_t myGREEN = display.color565(0, 255, 0);
uint16_t myBLUE = display.color565(0, 0, 255);

void display_updater()
{
  display.display(display_draw_time);
}

[... 299 lines cut ...]
      display.setTextColor(myRED);
      display.setCursor(17,17);
      display.print("Pixel");
      
    delay(3000);

      for (int i=0; i<128; i++) {
        for (int j=0; j<32; j++) {
          display.drawPixel(i,j, myRED);
          delay(10);
        }
      }
}

[https://www.youtube.com/watch?v=VKhJfgagdDA](https://www.youtube.com/watch?v=VKhJfgagdDA)

I think that, my problem is not about connections, cable lenght, ESP pins.
It's about R, G, B connection faulty. maybe last pin (B2) can connect to GND via 50 Ohm !!!!
or output pins not connect rightly. pin addresses faulty.

I dont know why.

Thank you @xsrf 
Thank you @2dom 
and thank you @rajeshdoshi

xsrf, 2020-05-21:
You should step back for a moment and check if your setup even works with one panel. It looks like it is not, groups of 8 rows each are actually in the wrong place. 
Can you just connect one panel and draw a white line from one corner to the other?
There is no need to terminate B2 output...
Also, please run the pattern_test from the examples with one panel and make a video.

adincergok, 2020-05-22:
Full original code of pattern_test:
#include <PxMatrix.h>

#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#endif
// Pins for LED MATRIX

PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);
//PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
//PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

#ifdef ESP8266
// ISR for display refresh
void display_updater()
{
  display.displayTestPattern(70);
  // display.displayTestPixel(70);
  //display.display(70);
}
#endif

#ifdef ESP32
void IRAM_ATTR display_updater(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  //isplay.display(70);
  display.displayTestPattern(70);
  portEXIT_CRITICAL_ISR(&timerMux);
}
#endif

uint16_t myCYAN = display.color565(0, 255, 255);
void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
  display.begin(4);
  display.flushDisplay();
  display.setTextColor(myCYAN);
  display.setCursor(2,0);
  display.print("Pixel");
  Serial.println("hello");

  #ifdef ESP8266
    display_ticker.attach(0.004, display_updater);
  #endif

  #ifdef ESP32
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &display_updater, true);
    timerAlarmWrite(timer, 4000, true);
    timerAlarmEnable(timer);
  #endif

  delay(1000);
}

void loop() {
 delay(100);
}

and video:  [https://www.youtube.com/watch?v=wZFEmJkcWAw](https://www.youtube.com/watch?v=wZFEmJkcWAw)

xsrf, 2020-05-22:
That doesn't look too bad. Your color order actually is `BBGGRR` so please try `setColorOrder(BBGGRR)` in regular code (it won't affect the pattern test).
Also, I assume the video was shot with `PxMATRIX display(32,32,P_LAT, P_OE,P_A,P_B,P_C);` and not 16 height, right?
When you let the example run longer, is it incrementally lighting up all rows from 0 to 7 the same way? For the first two it looks okay.

adincergok, 2020-05-22:
I change this (for my code running) 

yes 32,32  (i only run original code)
Now:
PxMATRIX display(32,32,P_LAT, P_OE,P_A,P_B,P_C);
.....
void display_updater()
{
  display.display(70);
}
.........
void loop() {
  display.clearDisplay();
  display.setBrightness(50);
  int x1=0, y1=0, x2=31, y2=31;
  display.drawLine(x1, y1, x2, y2, myCYAN);
 delay(1000);
}

[image]

adincergok, 2020-05-22:
sorry. added  BBGGRR

uint16_t myCYAN = display.color565(0, 255, 255);
void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
  display.begin(4);
  display.setColorOrder(BBGGRR);
  display.flushDisplay();
  display.setTextColor(myCYAN);
  display.setCursor(2,0);
  display.print("Pixel");
  Serial.println("hello");

  #ifdef ESP8266
    display_ticker.attach(0.004, display_updater);
  #endif

  #ifdef ESP32
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &display_updater, true);
    timerAlarmWrite(timer, 4000, true);
    timerAlarmEnable(timer);
  #endif

  delay(1000);
}

void loop() {
  display.clearDisplay();
  display.setBrightness(50);
  int x1=0, y1=0, x2=31, y2=31;
  display.drawLine(x1, y1, x2, y2, myCYAN);
 delay(1000);
}

[image]

adincergok, 2020-05-22:
I think I have to use

display.setPanelsWidth(2);

i think my panel is 32x16 size double panel inside !!!

xsrf, 2020-05-22:
why did you change to begin(4)? You must use begin(8).

Can you try this (untested)? Also, please put your code in code tags when you post here.

```
#include <PxMatrix.h>

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

PxMATRIX display(32,32,P_LAT, P_OE,P_A,P_B,P_C);

void display_updater() {
    display.display(70);
}

void setup() {
    display.begin(8);
    display.setPanelsWidth(2);
    display.setColorOrder(BBGGRR);
    display.flushDisplay();
    display_ticker.attach(0.004, display_updater);
    display.drawLine(0, 0, 31, 31, display.color565(255, 255, 255));
    display.drawLine(1, 0, 31, 30, display.color565(255, 0, 0));
    display.drawLine(2, 0, 31, 29, display.color565(0, 0, 255));
}

void loop() {
}
```

adincergok, 2020-05-22:
test your code. result:
[image]

xsrf, 2020-05-22:
Great... So everything is okay except that row 0-7 and 8-15 as well as 16-23 and 24-31 need to be swapped. I'm not sure if a suitable scanPattern exists. It is basically `LINE` but with top/bottom flipped. I'm not familial with the different scanPatternd. Any idea @2dom ?
@adincergok you may just use this code but try different scan patterns from `LINE, ZIGZAG, ZAGGIZ, WZAGZIG, VZAG`...

adincergok, 2020-05-22:
now I'm try scan patterns again. in 5 minutes.

adincergok, 2020-05-22:
scan pattern results:
[image]

xsrf, 2020-05-24:
Same as #198 ... needs implementation of a new scanPattern, I guess

adincergok, 2020-05-24:
@xsrf  you are right. library need a new scanPattern. is it posible? 
what's your advice? try to other library? i need your advice

adincergok, 2020-05-26:
I solve my problem. add this rows in PxMatrix.h 

inline void PxMATRIX::fillMatrixBuffer(int16_t x, int16_t y, uint8_t r, uint8_t g, uint8_t b,bool selected_buffer)
{
> **if (y<8)
> 	{
> 		y+=8;
> 	} else if (y<16) {
> 		y-=8;
> 	} else if (y<24) {
> 		y+=8;
> 	} else if (y<32) {
> 		y-=8;
> 	}**

  if (_rotate){
    uint16_t temp_x=x;
    x=y;
    y=_height-1-temp_x;
  }
.....................

8 rows block flipped. my problem solved.
But there is an other problem. I can connecte 4 panels (each panel is 32x32 px). When I connect 5th panel or 5 & 6th panels no any pixel light. All panel in darkness.

PxMATRIX display(160,32,P_LAT, P_OE,P_A,P_B,P_C);  // testing 5 panels
Or
PxMATRIX display(196,32,P_LAT, P_OE,P_A,P_B,P_C);   // testing 6 panels

no any image on panels. all blanks.

i think, width cannot be more than 128 pixels. is 128 limit?

2dom, 2020-06-12:
Have you tried WZAGZIG2 ?