2dom/PxMatrix #204: P4 40x80 Matrix

rajeshdoshi, 2020-05-20:
P4 modules are available now in 40x80 dots matrix. 
I tried to run with 
PxMATRIX display(80,40,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);              // Pins for LED MATRIX

However display remained blank. 
When I checked in PxMatrix.h file in library, I found Max width set as 64. So I made following changes in PxMatrix.h library

// Defines the buffer height / the maximum height of the matrix
#ifndef PxMATRIX_MAX_HEIGHT
#define PxMATRIX_MAX_HEIGHT 64
#endif

// Defines the buffer width / the maximum width of the matrix
#ifndef PxMATRIX_MAX_WIDTH
#define PxMATRIX_MAX_WIDTH 80
#endif

After making these changes, the display worked, but it was random. 
Does the library support 80x40 pixel module?

2dom, 2020-05-20:
Do you have a link to some documentation or product site?

rajeshdoshi, 2020-05-20:
No. These are Chinese slabs sold off the shelf or online. 
[image]

2dom, 2020-05-20:
Link?

rajeshdoshi, 2020-05-20:
http://www.qiangliled.com/products-19.html

2dom, 2020-05-20:
This looks like a pretty standard panel to me ... Are you sure it only has 80x40 pixels. The feature list is awesome though :)

rajeshdoshi, 2020-05-20:
As you can see from the web page link it is 320x160 mm size module ie 80 x40 dots. 
Will PxMatrix library work for this?

xsrf, 2020-05-20:
the library should work with 80x40 but the row-mapping might be different to regular binary decoding because 40/20/10 isn't a power of 2. The pinout in the middle shows that only A,B,C are used, right? Since the parameters on the website say it's 1/10 scan and 10 rows cannot be addressed with 3 bit I guess it uses a shift register for row addressing. Try `setMuxPattern(SHIFTREG)` and `begin(10)` as initial parameters.
If it doesn't work, show us a video of the pattern_test example with these settings.

2dom, 2020-05-20:
@rajeshdoshi ... Do you actually have this panel in front of you...may just be a typo in their product description. 40x80 would be truly odd

rajeshdoshi, 2020-05-21:
Yes I have these panels with me. I will connect them again to ESP later in the day and send you the results of the test. 
[image]

rajeshdoshi, 2020-05-21:
Ticker display_ticker;
PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);              // Pins for LED MATRIX
//PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);              // Pins for LED MATRIX
//PxMATRIX display(80,40,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);              // Pins for LED MATRIX

unsigned char i,j;

void setup() {
  Serial.begin(9600);

  display.begin(10);
//  display.setMuxPattern(SHIFTREG)

  display.flushDisplay();  
  display.setPanelsWidth(1);                          // Set the number of panels that make up the display area width (default is 1) can be 2 or more
  display.setBrightness(100);                         // Set the brightness of the panels (default is 255)
  display.clearDisplay();

  for(i=0;i<64;i++)
    for(j=0;j<32;j++)
      {
       display.drawPixel(i,j,Red);
       delay(100);
       display_updater_enable(true);
      }
  display_updater_enable(true);
 }

void loop() {
  // put your main code here, to run repeatedly:
  
}

display.setMuxPattern(SHIFTREG_ABC)  gives error. I am using PxMatrix 1.7 library. In .h file I found the accepted parameters to be STRAIGHT or BINARY. 

I have attached video in zip file. 

[p4_40x80.zip](https://github.com/2dom/PxMatrix/files/4661597/p4_40x80.zip)

rajeshdoshi, 2020-05-21:
Ticker display_ticker;

//PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);              // Pins for LED MATRIX
//PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);              // Pins for LED MATRIX
PxMATRIX display(80,40,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);              // Pins for LED MATRIX

uint16_t Red = display.color565(200, 0, 0);                       // Define colours for Day
uint16_t Green = display.color565(0, 60, 0);
uint16_t Blue = display.color565(0, 0, 120);
uint16_t White = display.color565(60, 60, 60);
uint16_t Yellow = display.color565(90, 60, 0);
uint16_t Cyan = display.color565(0, 60, 70);
uint16_t Magenta = display.color565(120, 20, 120);
uint16_t Orange = display.color565(150, 20, 5);
uint16_t Pink = display.color565(120, 25, 40);
uint16_t Iceblue = display.color565(20, 20, 100);
uint16_t Gold = display.color565(100, 20, 15);
uint16_t Black = display.color565(0, 0, 0);

void display_updater()
{
  display.display(70);
}

void display_updater_enable(bool is_enable)                           
{
if (is_enable)
    display_ticker.attach(0.002, display_updater);
  else
    display_ticker.detach();
}

unsigned char i,j;

void setup() {
  Serial.begin(9600);

  display.begin(10);
//  display.setMuxPattern(SHIFTREG)
  display.flushDisplay();  
  display.setPanelsWidth(1);                          // Set the number of panels that make up the display area width (default is 1) can be 2 or more
  display.setBrightness(100);                         // Set the brightness of the panels (default is 255)
  display.clearDisplay();

  for(i=0;i<64;i++)
    for(j=0;j<32;j++)
      {
       display.drawPixel(i,j,Red);
       delay(100);
       display_updater_enable(true);
      }
  display_updater_enable(true);
 }

void loop() {
  // put your main code here, to run repeatedly:
  
}

display.setMuxPattern(SHIFTREG_ABC) gives error. I am using Library V1.7.0 
I checked in PxMatrix.h file and it expects STRAIGHT OR BINARY as options. SHIFTREG_ABC is not there. 

Video is attached
[P4_40x80.zip](https://github.com/2dom/PxMatrix/files/4661630/P4_40x80.zip)

xsrf, 2020-05-21:
Can you run the pattern_test again and instead of connecting ABC to the micro-controller, just connect them to GND please. After you've done this, make sure to power off and power on the panel again before you start. If you see something, please make a video again. It may also be possible that it stays black this way.

rajeshdoshi, 2020-05-21:
Hey, I downloaded new library V1.8.1 and set Mux pattern to SHIFTREG_ABC and it worked!!!!

[P4_40x80.zip](https://github.com/2dom/PxMatrix/files/4661714/P4_40x80.zip)

Thank you very much !!! You guys rock !!

rajeshdoshi, 2020-05-21:
There is a continuous flicker which is not there when 64x32 display is used with same program. 
When display buffer is not written to, the display remains steady. But as soon as display buffer is written, there is a flicker.

Tried changing 
#define PxMATRIX_COLOR_DEPTH 4  as well as 3 and 2 as well
#define PxMATRIX_MAX_HEIGHT 80
#define PxMATRIX_MAX_WIDTH 40
#define PxMATRIX_SPI_FREQUENCY 10000000 as well as 10000000

closed 2020-05-26 by rajeshdoshi