2dom/PxMatrix #205: Using Double Buffer

rajeshdoshi, 2020-05-23:
I have query about double_buffer. How it works? 

As I understood, when we use draw commands, it is written to a buffer which is not displayed. When we call showBuffer, this buffer is transferred to display buffer. Is this correct? 

However, if that is the case, the display should appear transparent and without flicker. I have observed that the display goes blank for a fraction of a second when showBuffer is called. 

Is there something wrong in my understanding? I want to use double buffer so that there is no flicker when display data is updated.

2dom, 2020-05-23:
Yes...this is precisely how it works. Calling showBuffer should not cause any flicker since it only changes a pointer to the other frame buffer. Did you set #define PxMATRIX_DOUBLE_BUFFER  before including PxMatrix.h ?

rajeshdoshi, 2020-05-23:
Yes it is defined. Here is the code
I am using 3 slabs of 80x40 and hence my matrix is 240x40.
Here I am scrolling text. 

void loop() {
    display.setCursor(200,0);
    display.print("Awesome");
    display.showBuffer();
    scroll();
}

void scroll()
{
    do
    {
     x-=1;
    display.fillRect(x,22,239,37,Black);
    display.setCursor(x,22);
    display.println(Message[0]);
    display.showBuffer();
    delay(20);
    }while(x);
}

As you can see from video, I do not understand why should the top fix letters 'awesome' flash. Ideally they should remain steady. 

[scroll.zip](https://github.com/2dom/PxMatrix/files/4672097/scroll.zip)

xsrf, 2020-05-23:
You print `display.print("Awesome");` only to one framebuffer, because you swap the buffer twice inside the loop. The other framebuffer is always empty at this place.
Remove `display.showBuffer();` from either loop or scroll and it will work.

rajeshdoshi, 2020-05-24:
Is there a function which just copies one buffer to other? Or only way is to redraw on both the buffers?

xsrf, 2020-05-24:
You may try https://github.com/xsrf/PxMatrix/tree/copybuffer 
I've added `copyBuffer()` which you can execute after `showBuffer()` to copy the display buffer back to the drawing buffer.

rajeshdoshi, 2020-05-26:
Thanks. That is great. !!!
Closing this issue.

closed 2020-05-26 by rajeshdoshi