2dom/PxMatrix #206: Framebuffer size mismatch / 64 x 64 default

xsrf, 2020-05-24:
Hey,
I've just noticed that the size of the framebuffer(s) `PxMATRIX_buffer` / `PxMATRIX_buffer2` is calculated from `PxMATRIX_MAX_HEIGHT * PxMATRIX_MAX_WIDTH` which is usually 64 x 64 by default. Since the panel dimension is also defined in the constructor, that's kind of redundant and results in more memory being reserved than needed for panels smaller than 64 x 64.
Wouldn't it be better to eliminate `PxMATRIX_MAX_HEIGHT` and `PxMATRIX_MAX_WIDTH` and just use the values from the constructor?
This would mean that memory must be allocated dynamically, thus the compiler won't know the exact RAM used by variables though...

2dom, 2020-05-24:
Yes ... Very true. I think I opted for wasting some memory rather having potentially memory related crashes with novice users using to much ram (there might even be an old pull request somewhere). What do you think?

xsrf, 2020-05-24:
I think more users will have problems with the library not working correctly because they use e.g. two panels and haven't set `PxMATRIX_MAX_WIDTH` correctly.

The ESP8266 has about 60k of free RAM (with SDK but no user code) if I remember correctly. That would even fit a 512x256 panel with double buffering and 4 bit color depth without any problem.

Did you ever run into issues where the compiler told you that you're out of ram?

I used dynamic memory allocation on my own library, so I may give it a try... https://github.com/xsrf/ESP8266Matrix/blob/6a1aa1faaef6c914b32b7a57b692c10ad72c1ac2/ESP8266Matrix.h#L191

xsrf, 2020-05-24:
btw. is the library still tested on Atmel? The ATmega328P only has 2KB RAM, which is already exceeded by 64x64 if more than one bit color depth is used.

2dom, 2020-05-24:
Well ... this is why the AVR example defines 

#define PxMATRIX_COLOR_DEPTH 1
#define PxMATRIX_MAX_HEIGHT 32
#define PxMATRIX_MAX_WIDTH 64

but I think you are right. probably time to use dynamic mem allocation.

2dom, 2020-06-01:
OK ... can you check out the latest master. I moved the frame buffer and the row_offset data structures to dynamic memory.

xsrf, 2020-06-02:
I've looked at the changes of 6e1d301645d1d7cd97a0c038a8ec9d5b866f35a6 and also tested the code on an ESP8266 and ESP32. On the ESP8266 it worked fine, but on the ESP32 it won't compile because of https://github.com/2dom/PxMatrix/blob/7c403eef5ac3bc2f61ba8edd3a13708f058ae226/PxMatrix.h#L1161
Guess it should be 
```
uint8_t* bf = &PxMATRIX_bufferp[_display_color*_buffer_size+i*_send_buffer_size];
```

2dom, 2020-06-02:
Fixed ...

closed 2020-06-02 by xsrf