2dom/PxMatrix #209: 64x64 P3 Panel Setup

swraife, 2020-05-29:
I'm stumped on how to get this 64x64 P3 indoor  1/32 scan rate panel working. https://www.alibaba.com/product-detail/led-matrix-64x64-led-display-module_60829580307.html?spm=a2756.order-detail-ta-bn-b.0.0.58c02fc2m8tHah

I'm using a D1 mini clone with an ElectoDragon shield, described here: https://iot-for-maker.blogspot.com/2020/02/led-8-rgb-led-matrix-drive-with-esp.html . All switches are on as it's 1/32 scan rate. The chip and shield work fine on 32x32 1/8 and 1/16 panels that I have. I can't get pattern_test to work at all, so I've posted a video of pixel_time: https://www.youtube.com/watch?v=g0LAljyPcpI . The "Pixel Time" at the start of the program doesn't display at all, just what you see in the video. I've tried all the scan patterns, the 6124 chip (just displays vertical cyan lines), most all the variations I could come up with. The code below (what is in the video) seems closest. Also, if i just do drawPixel at [0,0] it displays at [0,0], [0,2], [0,4], and [0,6] (presumably anyway, as the corner pixel broke off). Colors are accurate including the blue which is missing from the pixel_time demo. 

`// This is how many color levels the display shows - the more the slower the update
//#define PxMATRIX_COLOR_DEPTH 8

// Defines the buffer height / the maximum height of the matrix
#define PxMATRIX_MAX_HEIGHT 64

// Defines the buffer width / the maximum width of the matrix
#define PxMATRIX_MAX_WIDTH 64
//
// Defines how long we display things by default
//#define PxMATRIX_DEFAULT_SHOWTIME 30

// Defines the speed of the SPI bus (reducing this may help if you experience noisy images)
//#define PxMATRIX_SPI_FREQUENCY 20000000

// Creates a second buffer for backround drawing (doubles the required RAM)
//#define PxMATRIX_double_buffer true

#include <PxMatrix.h>

// Pins for LED MATRIX
#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#endif

#define matrix_width 64
#define matrix_height 64

// This defines the 'on' time of the display is us. The larger this number,
// the brighter the display. If too large the ESP will crash
uint8_t display_draw_time=60; //30-70 is usually fine

//PxMATRIX display(32,32,P_LAT, P_OE,P_A,P_B,P_C);
//PxMATRIX display(32,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D, P_E);

// Some standard colors
uint16_t myRED = display.color565(255, 0, 0);
uint16_t myGREEN = display.color565(0, 255, 0);
uint16_t myBLUE = display.color565(0, 0, 255);
uint16_t myWHITE = display.color565(255, 255, 255);
uint16_t myYELLOW = display.color565(255, 255, 0);
uint16_t myCYAN = display.color565(0, 255, 255);
uint16_t myMAGENTA = display.color565(255, 0, 255);
uint16_t myBLACK = display.color565(0, 0, 0);

uint16_t myCOLORS[8]={myRED,myGREEN,myBLUE,myWHITE,myYELLOW,myCYAN,myMAGENTA,myBLACK};

uint8_t static weather_icons[]=
// Removed for readability

#ifdef ESP8266
// ISR for display refresh
void display_updater()
{
  display.display(display_draw_time);
}
#endif

#ifdef ESP32
void IRAM_ATTR display_updater(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  display.display(display_draw_time);
  portEXIT_CRITICAL_ISR(&timerMux);
}
#endif

void display_update_enable(bool is_enable)
{

#ifdef ESP8266
  if (is_enable)
    display_ticker.attach(0.004, display_updater);
  else
    display_ticker.detach();
#endif

#ifdef ESP32
  if (is_enable)
  {
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &display_updater, true);
    timerAlarmWrite(timer, 4000, true);
    timerAlarmEnable(timer);
  }
  else
  {
    timerDetachInterrupt(timer);
    timerAlarmDisable(timer);
  }
#endif
}

void setup() {

 Serial.begin(115200);
  // Define your display layout here, e.g. 1/8 step, and optional SPI pins begin(row_pattern, CLK, MOSI, MISO, SS)
  display.begin(32);
  //display.begin(8, 14, 13, 12, 4);

  display.print("my red:");
  display.print(myRED);
  // Define multiplex implemention here {BINARY, STRAIGHT} (default is BINARY)
//  display.setMuxPattern(BINARY);

  // Set the multiplex pattern {LINE, ZIGZAG,ZZAGG, ZAGGIZ, WZAGZIG, VZAG, ZAGZIG} (default is LINE)
//  display.setScanPattern(ZIGZAG);

  // Rotate display
  //display.setRotate(true);

  // Flip display
  //display.setFlip(true);

  // Control the minimum color values that result in an active pixel
  //display.setColorOffset(5, 5,5);

  // Set the multiplex implemention {BINARY, STRAIGHT} (default is BINARY)
//  display.setMuxPattern(STRAIGHT);

  // Set the color order {RRGGBB, RRBBGG, GGRRBB, GGBBRR, BBRRGG, BBGGRR} (default is RRGGBB)
  //display.setColorOrder(RRGGBB);

  // Set the time in microseconds that we pause after selecting each mux channel
  // (May help if some rows are missing / the mux chip is too slow)
//  display.setMuxDelay(0,1,0,0,0);

  // Set the number of panels that make up the display area width (default is 1)
  //display.setPanelsWidth(2);

  // Set the brightness of the panels (default is 255)
  //display.setBrightness(50);

  // Set driver chip type
//  display.setDriverChip(FM6124);

  display.clearDisplay();
  display.setTextColor(myCYAN);
  display.setCursor(2,0);
  display.print("Pixel");
  display.setTextColor(myMAGENTA);
  display.setCursor(2,8);
  display.print("Time");
  display_update_enable(true);

  delay(3000);

}
union single_double{
  uint8_t two[2];
  uint16_t one;
} this_single_double;

// This draws the weather icons
void draw_weather_icon (uint8_t icon)
{
  if (icon>10)
  icon=10;
  for (int yy=0; yy<10;yy++)
  {
    for (int xx=0; xx<10;xx++)
    {
      uint16_t byte_pos=(xx+icon*10)*2+yy*220;
      this_single_double.two[1]=weather_icons[byte_pos];
      this_single_double.two[0]=weather_icons[byte_pos+1];
      display.drawPixel(1+xx,yy,this_single_double.one);
    }
  }
}

unsigned long last_draw=0;
void scroll_text(uint8_t ypos, unsigned long scroll_delay, String text, uint8_t colorR, uint8_t colorG, uint8_t colorB)
{
    uint16_t text_length = text.length();
    display.setTextWrap(false);  // we don't wrap text so it scrolls nicely
    display.setTextSize(1);
    display.setRotation(0);
    display.setTextColor(display.color565(colorR,colorG,colorB));

    // Asuming 5 pixel average character width
    for (int xpos=matrix_width; xpos>-(matrix_width+text_length*5); xpos--)
    {
      display.setTextColor(display.color565(colorR,colorG,colorB));
      display.clearDisplay();
      display.setCursor(xpos,ypos);
      display.println(text);
      delay(scroll_delay);
      yield();

      // This might smooth the transition a bit if we go slow
      // display.setTextColor(display.color565(colorR/4,colorG/4,colorB/4));
      // display.setCursor(xpos-1,ypos);
      // display.println(text);

      delay(scroll_delay/5);
      yield();

    }
}

uint8_t icon_index=0;
void loop() {
  scroll_text(1,50,"Welcome to PxMatrix!",96,96,250);
  display.clearDisplay();

  draw_weather_icon(icon_index);
  icon_index++;
  if (icon_index>10)
    icon_index=0;

  for (int xx=0; xx<16;xx++)
  {
    display.drawLine(xx+16,0,xx+16,5,display.color565(xx*16,0,0));
    display.drawLine(xx+16,6,xx+16,10,display.color565(0,xx*16,0));
    display.drawLine(xx+16,11,xx+16,15,display.color565(0,0,xx*16));
  }
  delay(1000);
  for (uint8_t dimm=255; dimm>0; dimm--)
  {
    display.setBrightness(dimm);
    delay(5);
  }
  for (uint8_t dimm=0; dimm<255; dimm++)
  {
    display.setBrightness(dimm);
    delay(5);
  }

}`

xsrf, 2020-05-29:
have you tried `display.setMuxPattern(SHIFTREG_ABC);`?

I also worked on a debug sketch that may help to find out what parameters are needed for all kind of HUB75 panels. Maybe give it a try: https://github.com/xsrf/HUB75Debug/blob/master/src/hub75debug.cpp

swraife, 2020-05-29:
Thanks.  I downloaded master branch and loaded into the Arduino IDE to try it. Nothing shows up on the panel when I try either of the new SHIFTREG mux patterns. But I haven't tried all the combinations with scan pattern yet. I'll try your debugger too then report back.

swraife, 2020-05-30:
@xsrf I tried your debugger but couldn't get anything to display. I only changed the matrix dimensions as all the wiring looked correct. It worked with my 32x32 1/8 panel.

I also tried some more various other combos with the two new muxPatterns, but nothing shows up. Here is a pic of the back of the panel if that helps. Also, I said in my first post that 'Pixel Time' test doesn't appear, but actually it sorta does as this weird thing. That's using the code above. 
[image]
[image]

xsrf, 2020-05-30:
So, you got nothing on this panel with my debug code, but on an other one the debug code worked? hmm... can you give the part numbers of the ICs on the panel and how many of them there are?

swraife, 2020-05-30:
@xsrf Correct. Nothing displayed on this panel, but I got what I assumed to be correct on a different kind of panel with your debug code, just to ensure I hadn't screwed something up. 

Hard to say if there  are any underneath the fatter plastic running vertically, but I don't think so. The picture color codes which are which. Quantities: 
SM16237DS => 12
74HC245C => 4
SM5266PH => 8
74HC135T => 1
[image]

xsrf, 2020-05-30:
can you try my debug code, but with B and C swapped?

swraife, 2020-05-30:
I assume you mean switch the wiring in the code like so? If so, still nothing is displayed.

`// Use PIN_NC for unconnected Pins
#define PIN_NC 255

// Matrix Dimensions
#define MATRIX_WIDTH 64
#define MATRIX_HEIGHT 64

// PIN Wiring, set to PIN_NC if not connected
#define P_LAT  16 // D0
#define P_A     5 // D1
#define P_B    15 // D2
#define P_C    4 // D8
#define P_D    12 // D6 (Only for 1/16 or 1/32 panels)
#define P_E     0 // D3 (Only for 1/32 panles)
#define P_OE    2 // D4
#define P_CLK  14 // D5 (Documentation only, this Pin is hardwired and cannot be changed!)
#define P_DATA 13 // D7 (Documentation only, this Pin is hardwired and cannot be changed!)`

xsrf, 2020-05-30:
yep, hmmm... The SM5266PH is a shift register based LED driver... But it is not clear how it is wired to A-E pins. The silkscreen on your panel even has E instead of GND so it may also be connected somehow to this shiftregister. Is it possible for you to trace back the connections of A-E to this chip somehow?

swraife, 2020-05-30:
Oh jeez. Not sure what I can really tell. The manufacturer offered a control card debugging program? What specifically should I ask them for to help figure it out? 
[image]

xsrf, 2020-05-30:
Hmm, I tried to adapt my code so it can detect which PIN of A-D is the data input of the register, and which one is clock... You may try this: https://github.com/xsrf/HUB75Debug/blob/shiftreg-pindetect/src/hub75debug.cpp
If it works, one of the first 25 rows should light up.
I wasn't able to test the code though, I currently don't have access to any hardware.

swraife, 2020-05-30:
That works! 4th and 36th from the top light up.

xsrf, 2020-05-30:
Okay, so A is the Clock-Input and D is the Data-Input of the Shift-Register. SHIFTREG_ABC should work if you actually set the P_C variable to the pin that connects to D on the panel. e.g. swap P_C and P_D...

swraife, 2020-05-30:
No such luck. Tried pixel_time and pattern_test and nothing on either. To be sure I'm not missing anything, the changes I made on the pixel_time example were: 
`#define PxMATRIX_MAX_HEIGHT 64
 #define PxMATRIX_MAX_WIDTH 64
#define matrix_width 64
#define matrix_height 64

PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

void setup() {
  display.begin(32);
  display.setMuxPattern(SHIFTREG_ABC);
  ... `

xsrf, 2020-05-30:
Yeah, I guess I got my own debug code wrong 🙈 It should be B for data, not D... But haven't you already tried swapping B and C? 🤔

I guess it's easier if you keep all #define labeled correctly and just swap the order in the constructor:

`PxMATRIX display(64, 64, P_LAT, P_OE, P_A, P_C, P_B, P_D, P_E);`

swraife, 2020-05-30:
I did on your debugger code, which didn't work, but I hadn't tested on pixel_time I guess. That almost works! 'Pixel' displays at the start, but not 'Time' below it. The 'Welcome to PxMatrix!' scrolls fine. Then the weather icons show and and all the red box displays, but then only two rows of green. Basically it seems like only the first 8 rows are displaying.

xsrf, 2020-05-31:
Can you draw a white line from 0,0 to 31,63 and show a picture of it?

swraife, 2020-05-31:
[image]

swraife, 2020-05-31:
Looping through with drawPixel to light the entire board lights up everything in rows 1-8 and rows 33-40, as you would probably guess it would from the line photo above.

xsrf, 2020-05-31:
so, for whatever reason, only the first set of SM5266PH is active ... again, without an actual wiring diagram that's really difficult to debug. Don't have an idea at the moment... I assume there is some additional logic behind the pins C-E but I have no clue at all why a manufacturer would do that...

xsrf, 2020-05-31:
74HC135T is a binary 3 to 8 decoder. Maybe C-D are inputs to it and select which of the SM5266PH is active...
Can you add `digitalWrite(P_C, HIGH);` (and try also with D/E) after `display.setMuxPattern(SHIFTREG_ABC);` with the same code and report if/what changes?

If you can, just make sure with a multimeter that actually none of the C-E Pins is directly connected to GND. Your ESP wouldn't be happy setting a PIN to HIGH if it is connected to GND.

https://hackaday.io/project/170109-improved-raspberry-pi-rgb-matrix-driver/log/174236-qiangli-q3f32-v41 actually suggests it's a mix of binary decoding and shift register, which would be completely stupid?!

swraife, 2020-05-31:
C is exactly the same. D is the same but everything is shifted down 8 rows. E has everything shifted down 16 rows.

swraife, 2020-05-31:
Also, any suggestions for an aliexpress company that consistently sells not stupidly set up 64x64 indoor panels?

xsrf, 2020-05-31:
and D and E together? Shifted by 24?
Sorry, I can't get shopping advice...

swraife, 2020-05-31:
Yep. Shifted by 24.

xsrf, 2020-05-31:
Can you try https://github.com/xsrf/PxMatrix/tree/shiftreg-binary with
```
PxMATRIX display(64, 64, P_LAT, P_OE, P_A, P_B, P_C, P_D, P_E);
void setup() {
  display.begin(32);
  display.setMuxPattern(SHIFTREG_ABC_BIN_DE);
```
It should work ...

swraife, 2020-05-31:
It does! Thank you!

closed 2020-05-31 by swraife

xsrf, 2020-05-31:
I made a PR to add this to the library...
I still can't get my head around why someone would design a panel like this. Controlling it requires custom made software and hardware wise it is just more expensive. I don't see any benefits...

2dom, 2020-06-01:
Most of those scanning patterns Don't make much sense either...I think they have challenges in China who comes up with the craziest panel possible ...

lefty01, 2021-06-11:
feeling a bit dump right now but how to wire the R1/2,G1/2, and B1/2 PINS for such panel (64x64)?
I could not find  what the expected default would be nor where to change those if my wiring differs while  using the PxMatrix  code.

xsrf, 2021-06-11:
@lefty01 you connect the R1 output to the R2 input, the R2 output to the G1 input, the G1 output to the G2 input and so on. In the end, all RGB channels are chained together. The data is fed into R1 from your controller.

lefty01, 2021-06-11:
> @lefty01 you connect the R1 output to the R2 input, the R2 output to the G1 input, the G1 output to the G2 input and so on. In the end, all RGB channels are chained together. The data is fed into R1 from your controller.

oh... I meant the connection from the controller ;) for example I have a  64x64 panel hard-wired like this:
```
#define R1_PIN_DEFAULT  25
#define G1_PIN_DEFAULT  26
#define B1_PIN_DEFAULT  27
#define R2_PIN_DEFAULT  14
#define G2_PIN_DEFAULT  12
#define B2_PIN_DEFAULT  13

#define A_PIN_DEFAULT   23
#define B_PIN_DEFAULT   19
#define C_PIN_DEFAULT   5
#define D_PIN_DEFAULT   17
#define E_PIN_DEFAULT   32 // IMPORTANT: Change to a valid pin if using a 64x64px panel.
          
#define LAT_PIN_DEFAULT 4
#define OE_PIN_DEFAULT  15
#define CLK_PIN_DEFAULT 16

```
how does PxMatrix know the pin assigmnet for R,G, and B?

I have the panel working with the SmartMAtrix and ESP32-HUB75-MatrixPanel-I2S-DMA library and there I can find the pinout in some header file but for PxMatrix I just saw "configs" for the A,B,C,D,E, as well as clk and enable ... 
so just curious with this lib and a given 64x64 1/32 scan panel what is the expected default pins for an ESP32 controller. At least I was under the assumption that 64x64 matric panels are supported.
thx for your quick reply to the closed issues anyways ;) 👍

2dom, 2021-06-11:
As @xsrf said...PxMatrix only uses a single pin for serialised display data out put (SPI)

lefty01, 2021-06-12:
ah sorry ... now after reading again it makes sense, misunderstanding on my side! thx!

fragmation, 2023-02-26:
as I see and used this pxmatrix code and no work over 35 times, and joining lefty01, how can work a panel P3 without the r,g,b signals ? with the same ESP32-HUB75-MatrixPanel-I2S-DMA I can see draws with bitmap2hex conversor and putting it on the code, but with your pxmatrix cannot make it work in anyway, any solution ?