2dom/PxMatrix #210: Flickering while trying to do multiple lines of scrolling text

alfo, 2020-05-30:
I should start off by saying that if I use the included pixel_time scrolling text example, everything works fine, the problem is my code below and I was hoping someone could help me optimise it to remove the flickering.

I am trying to write a program that has some static text one each line, and some scrolling text that doesn't overlap with the scrolling. Below is a video of what I have working at the moment.

Here's my code:

```c
#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#define matrix_width 64
#define matrix_height 32

#include <PxMatrix.h>      // To drive the Matrix

// This defines the 'on' time of the display is us. The larger this number,
// the brighter the display. If too large the ESP will crash
uint8_t display_draw_time=60; //30-70 is usually fine

PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);

void IRAM_ATTR display_updater(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  display.display(display_draw_time);
  portEXIT_CRITICAL_ISR(&timerMux);
}

void display_update_enable(bool is_enable) {
  if (is_enable) {
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &display_updater, true);
    timerAlarmWrite(timer, 4000, true);
    timerAlarmEnable(timer);
  } else {
    timerDetachInterrupt(timer);
    timerAlarmDisable(timer);
  }
}

void fillBlankRow(uint8_t ypos) {

  // Clear only the pixels on this part of the display
  for (int x = 0; x < matrix_width; x++) {
    for (int y = ypos; y < ypos + 8; y++) {
      display.drawPixel(x, y, display.color565(0, 0, 0 ));
    }
  }
  
}

void drawStaticAndScrollingText(uint8_t ypos, unsigned long scroll_delay, String staticText, String scrollingText, uint8_t colorR_a, uint8_t colorG_a, uint8_t colorB_a, uint8_t colorR_b, uint8_t colorG_b, uint8_t colorB_b) {

    uint8_t characterWidth = 5;

    uint16_t scrollingTextLength = scrollingText.length();
    uint16_t offset = staticText.length() * 6;
    display.setTextWrap(false);
    display.setTextSize(1);
    display.setRotation(0);

    // Asuming 5 pixel average character width
    for (int xpos = offset; xpos > -(matrix_width + scrollingTextLength * characterWidth) + 50; xpos--) {

      display_update_enable(false);

      // Clear only the pixels on this part of the display
      fillBlankRow(ypos);

      //display_update_enable(true);
      //display_update_enable(false);

      display.setTextColor(display.color565(colorR_b,colorG_b,colorB_b));
      display.setCursor(xpos, ypos);
      display.println(scrollingText);

      for (int x = 0; x < offset; x++) {
        for (int y = ypos; y < ypos + 8; y++) {
          display.drawPixel(x, y, display.color565(0, 0, 255));
        }
      }

      display.setCursor(0, ypos);
      display.setTextColor(display.color565(colorR_a,colorG_a,colorB_a));
      display.println(staticText);

      display_update_enable(true);
      
      delay(scroll_delay);
      yield();

    }

    display.setTextColor(display.color565(colorR_b,colorG_b,colorB_b));
    display.setCursor(offset, ypos);
    display.println(scrollingText);

    
}

void setup() {
  // put your setup code here, to run once:
  display.begin(16);
  display.setFastUpdate(true);

  // Required for this model of display
  display.setMuxDelay(0,2,0,0,0);

  display.clearDisplay();
  display_update_enable(true);

}

void loop() {
  // put your main code here, to run repeatedly:

  drawStaticAndScrollingText(0, 50, "1a", "First Scrolling Text", 255, 80, 0, 150, 150, 150);
  delay(1000);
  drawStaticAndScrollingText(8, 50, "2a", "Second Scrolling Text", 255, 80, 0, 150, 150, 150);
  delay(1000);
  drawStaticAndScrollingText(16, 50, "3a", "Third Scrolling Text", 255, 80, 0, 150, 150, 150);
  delay(1000);
  drawStaticAndScrollingText(24, 50, "4a", "Final Scrolling Text", 255, 80, 0, 150, 150, 150);
  delay(5000);

}
```

As you can see, I've tried to use `display_update_enable(false)` so that the "background" pixels get set and the foreground text pixels get set *before* sending the update to the display to reduce flicker that way, but it hasn't worked.

I can't use the `clearDisplay()` function like the example because I don't want to get rid of what's on the other rows of the display, so I'm resorting to manually setting the pixels to be black in a loop, which I suspect isn't very efficient.

Does anyone have any pointers?

alfo, 2020-05-30:
Video here: [scroll_video.zip](https://github.com/2dom/PxMatrix/files/4705655/scroll_video.zip)

xsrf, 2020-05-30:
Disabling the update won't help you. It just causes the display to stay off for longer than the usual period, which will also result in flicker.
You may have a look at the double buffering example... have you tried using double buffering?
Be aware that with double buffering you have to either draw your static text to both buffers or copy the display buffer over to the drawing buffer before you draw your next frame using `display.copyBuffer()`. Otherwise your static text may flicker because it is only drawn every other frame.

alfo, 2020-05-30:
I haven't tried double buffering, but I'll look into it. I'm having a hard time understanding how to choose which buffer to write to, though, and how to tell which buffer is being displayed/next to be displayed?

xsrf, 2020-05-30:
you don't really have to worry about that. Just call `display.showBuffer()` when you've done everything to draw one full frame. As said, this may cause issues with every other frame. If it does, add `display.copyBuffer();` right after showBuffer.

RobertHolscher, 2020-05-30:
Hi 

I am struggling with the same. Also I see flickering when opening and reading from spiffs. 

Maybe it is an idea to index the buffer and write to that index and showbuffer[0] and showbuffer[1] and write to index 0 or 1 

This gives more flexibility in the coding 

Kind regards 

Robert 

Verstuurd vanaf mijn iPhone

alfo, 2020-05-30:
> add `display.copyBuffer();` right after showBuffer.

Weirdly, I'm getting a `'class PxMATRIX has no member named 'copyBuffer'` even though my sketch has:

```c
#define double_buffer true
#include <PxMatrix.h>
```

Looking at the source, this should work?

xsrf, 2020-05-30:
Oops, it has been added to the master branch just yesterday but it's not part of a new release yet. For now you may have to add the latest PxMatrix.h manually to your project.

alfo, 2020-05-30:
Aha I just figured that out you replied. This was a much more simple fix than I thought, thank you for the pointers! Below is my new working (non-flickering) code:

```c
#define double_buffer true
#include <PxMatrix.h>

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#define matrix_width 64
#define matrix_height 32

// Pins for LED MATRIX
PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);

void IRAM_ATTR display_updater(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  display.display(70);
  portEXIT_CRITICAL_ISR(&timerMux);
}

void fillBlankRow(uint8_t ypos) {

  // Clear only the pixels on this part of the display
  for (int x = 0; x < matrix_width; x++) {
    for (int y = ypos; y < ypos + 8; y++) {
      display.drawPixel(x, y, display.color565(0, 0, 0 ));
    }
  }
  
}

void drawStaticAndScrollingText(uint8_t ypos, unsigned long scroll_delay, String staticText, String scrollingText, uint8_t colorR_a, uint8_t colorG_a, uint8_t colorB_a, uint8_t colorR_b, uint8_t colorG_b, uint8_t colorB_b) {

    uint8_t characterWidth = 5;

    uint16_t scrollingTextLength = scrollingText.length();
    uint16_t offset = staticText.length() * 6;
    display.setTextWrap(false);
    display.setTextSize(1);
    display.setRotation(0);

    // Asuming 5 pixel average character width
    for (int xpos = offset; xpos > -(matrix_width + scrollingTextLength * characterWidth) + 50; xpos--) {

      // Clear only the pixels on this part of the display
      fillBlankRow(ypos);

      //display_update_enable(true);
      //display_update_enable(false);

      display.setTextColor(display.color565(colorR_b,colorG_b,colorB_b));
      display.setCursor(xpos, ypos);
      display.println(scrollingText);

      for (int x = 0; x < offset; x++) {
        for (int y = ypos; y < ypos + 8; y++) {
          display.drawPixel(x, y, display.color565(0, 0, 255));
        }
      }

      display.setCursor(0, ypos);
      display.setTextColor(display.color565(colorR_a,colorG_a,colorB_a));
      display.println(staticText);

      display.showBuffer();
      display.copyBuffer();
      delay(scroll_delay);
      yield();

    }

    display.setTextColor(display.color565(colorR_b,colorG_b,colorB_b));
    display.setCursor(offset, ypos);
    display.println(scrollingText);

    
}

void setup() {
  // put your setup code here, to run once:
  display.begin(16);

  // Required for this model of display
  display.setMuxDelay(0,2,0,0,0);
  
  display.flushDisplay();
  display.setTextWrap(false);

  timer = timerBegin(0, 80, true);
  timerAttachInterrupt(timer, &display_updater, true);
  timerAlarmWrite(timer, 4000, true);
  timerAlarmEnable(timer);
  
}

void loop() {
  // put your main code here, to run repeatedly:

  drawStaticAndScrollingText(0, 50, "1a", "First Scrolling Text", 255, 80, 0, 150, 150, 150);
  delay(1000);
  drawStaticAndScrollingText(8, 50, "2a", "Second Scrolling Text", 255, 80, 0, 150, 150, 150);
  delay(1000);
  drawStaticAndScrollingText(16, 50, "3a", "Third Scrolling Text", 255, 80, 0, 150, 150, 150);
  delay(1000);
  drawStaticAndScrollingText(24, 50, "4a", "Final Scrolling Text", 255, 80, 0, 150, 150, 150);
  delay(5000);

}
```

closed 2020-06-01 by alfo