2dom/PxMatrix #213: How to move the display functions to core 0 on an ESP32?

lg007, 2020-06-01:
Hello,

I have managed to put together a program that does the following with this excellent library:

- Drives a 64 x 32 RGB matrix,
- Connects to an MQTT Broker through a SIM800L GPRS modem,
- Displays a 3 digit number that it receives through MQTT.

Everything works well, but the code still needs heavy refactoring (it is really "ugly").

One thing I was planning is to move the display routines to core 0 on the ESP32 while leaving the MQTT and GPRS communication on core 1. I am not a professional programmer, so I would welcome any help on how to achieve this. Could you please help me?

Thanks,

mobby6kl, 2020-06-03:
Hi! Are you sure you need to do this? I have some pretty heavy stuff running (receiving stream of websocket data or downloading and parsing jsons while serving a web page) while drawing the screen at 80hz and didn't have to explicitly manage the affinity because of performance reasons.

The function to run a taks on a specific core is 
xTaskCreatePinnedToCore(yourFunction, "TaskName", 2048, NULL, 5, NULL, 0);
where yourFunction is the name of the function, the last 0 is the core. https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html

But there are some requiremetns, like the task can't return (has to run in a loop) and it can introduce other complications with synchronization with other tasks and data sharing, which is why I'm not sure it would be worth it if you're just drawing a 3 digit number on a small panel.

lg007, 2020-06-05:
Thanks, I managed to do it in the meawhile. But your comments are right, Core 0 is not needed for this simple task.

closed 2020-06-05 by lg007