2dom/PxMatrix #214: 64x64 P3 1/32 panel ABCE wiring

mobby6kl, 2020-06-01:
I've used PxMatrix successfully with different 64x32 panels but I couldn't get a newer 64x64 panel to work properly. It looks identical to the panel linked in issue #209 but completely different than the one swraife actually seemed to have so I was going to create a new issue.

I wrote up a detailed report but before submititng it, went to have another look to make sure I don't end up looking stupid. And I was in fact able to resolve the issue myself eventually. I'm not sure what the etiquet here is, but hopefully this can save someone a day frustration. Feel free to close it or I can do it.

Here's the planel: https://www.aliexpress.com/item/32959774837.html?spm=a2g0s.9042311.0.0.27424c4deqBKPB

[image]

[image]

[image]

That GND on pin 14? That's no ground at all, and is actually connected to pin 3 of the 74hc245 octal bus transceiver, and if set up as the D pin, everything works correctly. So be careful and don't trust the silkscreens, they can be flat out wrong. I wasted a lot of time trying permutations of scan and mux patterns for no reason.

So with this I also successfully set up two of them chained together for a 128x64 display and it works great. However there are a few minor things I ran into:

- Pin 2 on the ESP32 is a boot strapping pin and so it won't boot into programming mode if it's hooked up to a powered on display. Would it maybe make sense to change the default pin used in the guides and examples?
[image]
- The ESP32 crashes if I set the 1/32 scan in pixeltiem or double buffer examples and change nothing else. Changing display.display(40) or 30 works fine.
- The display.fillScreen(myBLACK) function in doublebuffer seems to be **very** slow. After I connected the second panel, the FPS went from 53 to 13 (so quarter performance with 2x pixels...). I guess it's a GFX function and I don't know what exactly it's doing, but the memset in clearDisplay() is at least an order of magnitude faster. It can also help to run the drawing code on another core.

2dom, 2020-06-01:
Hi @mobby6kl ... Thanks for the feedback. Yes...I should really get those things fixed as I have been postoning these changes for a while now. Will look into it soon ok :)

2dom, 2020-06-02:
OK ... 

- updated the Readme and the examples to use pin 16 instead of pin 2 throughout
- Updated examples to use a maximum of 20us in display.display()
- The difference in execution time for fillScreen and clearDisplay is due to the fact that fillScreen needs to compute the position of every pixel in the buffer. clearDisplay simply sets everything to zero. I think it is possible to provide a faster fillScreen function for every color but it is a bit more involved.

The current master also moved to dynamic memory allocation - can you run some examples and check if it works ok?

mobby6kl, 2020-06-03:
Great, thanks! Both double buffer and PixelTime examples work out of the box now (I was even using pin 16 for OE already) just configured for two 64x64, 1/32 displays. My existing code also seems to work fine with the new version.

The difference between fillScreen and clearDisplay makes sense. I don't think there's any need to re-implement the function, certainly not on my behalf! I just wanted to point out in case anyone runs into unexpected performance issues. I suppose fillScreen calls fillMatrixBuffer for every pixel? Looking at it, it's quite complex and also answeres a question I had about the framebuffer layout - "it's complicated" :)

By the way, have you noticed any performance impact in practice when using non-default SPI pins on the ESP32? The docs say changing them
> Increases the input delay of the MISO signal, which makes MISO setup time violations more likely. If SPI needs to operate at high speeds, use dedicated IO_MUX pins.
> Allows signals with clock frequencies only up to 40 MHz, as opposed to 80 MHz if IO_MUX pins are used.

The 40MHz frequency shouldn't be an issue if we're staying at 20 but no idea if the delays could be. I haven't had any problems in few tests I did, but just in case you or someone tried it with larger displays already.

In any case, thanks very much for this great library!

xsrf, 2020-06-03:
If I don't miss something, delays on the SPI bus are only relevant for bidirectional communication, because data is read while data is sent out. This is why the docs refer to MISO (Master In Slave Out = SPI Input). We are just sending data out to the displays. We don't really care if the data arrives delayed or not as long as it arrived before the LEDs get turned on ;)

closed 2020-06-12 by 2dom