2dom/PxMatrix #216: Slow Scrolling with multiple Slabs

rajeshdoshi, 2020-06-06:
I have used 5 slabs with one ESP controller. 
I am using your standard scrolling routine, modified a little as below. 

void scroll(int x)
{
    do
    {
     x-=1;
    display.fillRect(x,22,(80*SLABS)-1,37,Black);
    display.setCursor(x,22);
    display.println(Message[0]);
    display.showBuffer();
    display.copyBuffer();
    delay(1);
    yield();
    }while(x);
}

Attached video of scrolling. 
Is there any better way to make it scroll faster ? 

I thought of using multiple controllers to scan. I am thinking of using an interrupt when character reaches boundary of the slab and then the next slab will start the scroll further. Any suggestion? 

[scroll.zip](https://github.com/2dom/PxMatrix/files/4739663/scroll.zip)

rajeshdoshi, 2020-06-07:
I added a miils function to check how much time it takes. I found this routine takes 2500 miliseconds to complete one shift for 5 number of slabs. This I am checking on ESP8266.
Of these, 95% time is taken by fillRect routine. It takes about 2450 mS to fill rect. Is there any better option for this? 

void scroll(int x)
{
do
{
sprintf(buff,"Mills:%ld",mills());
Serial.println(buff);
x-=1;
display.fillRect(x,22,(80*SLABS)-1,37,Black);
sprintf(buff,"Mills:%ld",mills());
Serial.println(buff);
display.setCursor(x,22);
display.println(Message[0]);
display.showBuffer();
display.copyBuffer();
delay(1);
yield();
}while(x);
}

rajeshdoshi, 2020-06-07:
Is there a way where in I can fill the buffer memory directly which might be faster that fillRect?

xsrf, 2020-06-07:
So... this is an `Adafruit_GFX` issue, not an issue with `PxMatrix`. But since, as you say, `fillRect` takes longer than `println`, try to clear previous text by overwriting it with black instead of `fillrect`:
```
display.setTextColor(0x0000);
display.setCursor(x,22);
display.println(Message[0]);
x-=1;
display.setTextColor(0xFFFF);
display.setCursor(x,22);
display.println(Message[0]);
```

rajeshdoshi, 2020-06-07:
I will try that. Meantime, I tried clearDisplay with double_buffer enabled. 
I have two lines displayed. One is steady and other is scrolling. 
I observe that there is significant flicker in the steady line when other line starts to scroll. Ideally if showDisplay only changes the memory pointer, there should be no flicker. What can the reason be?

xsrf, 2020-06-07:
Probably your static text is only in one of the two buffers

rajeshdoshi, 2020-06-07:
After every showBuffer, I have called copyBuffer.

rajeshdoshi, 2020-06-07:
May be it is because of supply ripple when scrolling starts. May be I should check with additional filter caps across supply.

rajeshdoshi, 2020-06-07:
This is brilliant idea. It works much faster than FillRect. Aweseom !!

rajeshdoshi, 2020-06-07:
Hmm.. I got a very basic problem here. I am trying to calculate the total number of dots required in a line. So I am trying to access xAdvance value from the GFXglyph structure of the font. 
So I did something like this

const GFXfont * font_ptr;

In my function, to find number of dots required for the message, I do as follows. 

int a,z;

font_ptr = &FreeSerif9pt7b ;
wi=Message[0].length();
z=0;
for(a=0;a<wi;a++)
      z+=font_ptr->glyph[Message[0][a]-' '].xAdvance;
  
The core dumps at this instruction. 

Exception (3):<CR><LF>
epc1=0x40201ee4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4023b0c8 depc=0x00000000<CR><LF>
<CR><LF>
>>>stack>>><CR><LF>
<CR><LF>
ctx: cont<CR><LF>
sp: 3ffffdb0 end: 3fffffc0 offset: 0190<CR><LF>
3fffff40:  3ffee355 00000000 3ffee320 40201ed8  <CR><LF>
3fffff50:  3ffee355 00000036 3ffee390 3fffa268  <CR><LF>
3fffff60:  3fffa278 3ffee390 3ffee355 3fffa268  <CR><LF>
3fffff70:  3fffa278 3ffee390 3ffee355 402026c1  <CR><LF>
3fffff80:  00000000 00000000 00000001 4010017c  <CR><LF>
3fffff90:  3fffdad0 00000000 3fffa394 3fffa3d4  <CR><LF>
3fffffa0:  3fffdad0 00000000 3fffa394 40204788  <CR><LF>
3fffffb0:  feefeffe feefeffe 3ffe84ec 40100c95  <CR><LF>
<<<stack<<<<CR><LF>

I am not able to make out what is wrong in my code.

2dom, 2020-06-12:
Sorry ... no idea. Maybe check with the AdafruitGFX guys?

closed 2020-06-12 by 2dom