2dom/PxMatrix #218: New Panel

rajeshdoshi, 2020-06-10:
Hi, 
Back with another new panel from Quangli. This is 40x80 panel Q4Y10V6H.

Photograph of the panel is as here. 
[image]

Initialisation code I tried is

void setup() {
  int i=0,j=0;

  Serial.begin(9600);

  display.begin(10);
  display.setMuxPattern(BINARY);

  display.flushDisplay();  
  display.setPanelsWidth(1);                          
  display.setBrightness(100);                         
  display.clearDisplay();

  for(i=0;i<80;i++)
    for(j=0;j<40;j++)
      {
       display.drawPixel(i,j,Red);
       delay(10);
       display_updater_enable(true);
      }
}

The result is as shown in video...
From the part number, it seems 1/10 scan panel. Could not find its exact data sheet from their site though. On board they have shown D pin as well. So little confused. Can you guide please? 

[dotmx.zip](https://github.com/2dom/PxMatrix/files/4757634/dotmx.zip)

rajeshdoshi, 2020-06-11:
I tried making following change

#define MATRIX_WIDTH  80
#define MATRIX_HEIGHT 40

The display remains blank. Then I tried with other panel which works with SHIFT_ABC Mux pattern. Even that panel remains blank. 
Then I tried with 64x32 dot matrix, of course after changing Width and height, it shows white line.

xsrf, 2020-06-11:
I guess it may use a shiftregister with different A-D pinout. Can you note down the ICs used on the panel?

rajeshdoshi, 2020-06-11:
It is using SM5266 and 160206 display driver.

rajeshdoshi, 2020-06-11:
[SM5266P-SUNMOON.pdf](https://github.com/2dom/PxMatrix/files/4764124/SM5266P-SUNMOON.pdf)

rajeshdoshi, 2020-06-11:
[image]

xsrf, 2020-06-11:
You may try https://github.com/xsrf/PxMatrix/tree/shiftreg-binary with `SHIFTREG_ABC_BIN_DE` which is not yet merged into PxMatrix. It should work with 1/10 scanning if the first 8 rows are connected to the first SM5266 and the remaining two are connected to a second SM5266. But if they use only 5 of the 8 outputs of each SM5266, only 7 of 10 rows will work...

rajeshdoshi, 2020-06-11:
Great !!! Display started to work. It seems need to use different SCAN pattern 

for(i=0;i<80;i++)
for(j=0;j<40;j++)
{
display.drawPixel(i,j,Red);
delay(10);
}
}

I tried Line and Zigzag. With Zigzag following is the result  video. 
[q80.zip](https://github.com/2dom/PxMatrix/files/4764310/q80.zip)

xsrf, 2020-06-11:
please use `display.displayTestPixel(20);` instead of `display.display()` in your code and post a video to show how the pattern looks like.

rajeshdoshi, 2020-06-11:
[vid80.zip](https://github.com/2dom/PxMatrix/files/4764566/vid80.zip)
Please find attached. Thanks..

xsrf, 2020-06-11:
not sure if a pattern exists for that... @2dom ?

2dom, 2020-06-11:
Nope ... appears to be a byte split pattern. I already implemented another one - should not be a biggie.

2dom, 2020-06-11:
Would be ZZIAGG I suppose :)

rajeshdoshi, 2020-06-11:
Hi...by the way I had made a small donation on Monday. Hope you received it. Thanks.

2dom, 2020-06-11:
Thanks man ... In this case all the credit goes to XSRF, @xsrf - if you PM me your PayPal very happy to forward you some beer money :)

xsrf, 2020-06-11:
Thx, but please keep it 😊 However, I wouldn't mind if you could add some visuals of the different patterns to the readme some time 😉

2dom, 2020-06-11:
Yeah ... I should probably do that. I'm not entirely sure if the whole naming thing still makes sense - this is getting out of hand ...

2dom, 2020-06-11:
@rajeshdoshi I implemented a new pattern ZZIAGG - check out the latest master if it does anything useful.

rajeshdoshi, 2020-06-11:
Oh great. Thanks.. will check and let you know.

rajeshdoshi, 2020-06-11:
Hope it includes SHIFTREG_ABC_BIN_DE  as well.

rajeshdoshi, 2020-06-11:
Some thing still missing. Here is video... It is supposed to fill in dots from left corner top to bottom. 
[vid.zip](https://github.com/2dom/PxMatrix/files/4765818/vid.zip)

2dom, 2020-06-11:
Ok ... could you try again with the latest master and

 ```
for (int y=0; y<40; y++)
    {
        for (int x=0; x<80; x++)
        {
            display.drawPixel(x, y, myCYAN);
  
            delay(20);   
        }
    }
```

rajeshdoshi, 2020-06-12:
Almost there....
Here is video...

[vid1.zip](https://github.com/2dom/PxMatrix/files/4768564/vid1.zip)

2dom, 2020-06-12:
Could you try again with the latest master?

rajeshdoshi, 2020-06-13:
Opps. I put my comments in wrong place. 
I tried with new master.  Still some problem..

[Uploading vid_1.zip…]()

rajeshdoshi, 2020-06-13:
[vid_1.zip](https://github.com/2dom/PxMatrix/files/4774028/vid_1.zip)

xsrf, 2020-06-13:
The mux pattern is not correct for your panel. Can you please post the complete code?

rajeshdoshi, 2020-06-14:
void setup() {
 int i=0,j=0;

  Serial.begin(115200);
  Serial.println("Esp Started");
  display.begin(10);
  display.setMuxPattern(SHIFTREG_ABC_BIN_DE);
  display.setScanPattern(ZZIAGG);

  display.flushDisplay();  
  display.setPanelsWidth(1);                          // Set the number of panels that make up the display area width (default is 1) can be 2 or more
  display.setBrightness(250);                         // Set the brightness of the panels (default is 255)
  display.clearDisplay();

display_updater_enable(true);
for (int y=0; y<40; y++)
   {
       for (int x=0; x<80; x++)
       {
           display.drawPixel(x, y, Cyan);
           delay(20);   
       }
   }
}

xsrf, 2020-06-14:
can you please check your wiring of Pin D? You already had all rows working... Now it looks like it is stuck on the 2nd pair of rows.

rajeshdoshi, 2020-06-15:
Problem is not with D pin. 
I loaded earlier commit
https://github.com/2dom/PxMatrix/blob/d120e24fd9fdc6cad7c84c8f89084ff9a94e0972/PxMatrix.h
And the result was as per vid1.zip I had shared earlier. 
[image]

2dom, 2020-06-15:
I am confused ... it should scan x from 1 to 80 first, now goes to to bottom. Everything worked in the first set of rows in your "almost there" comment. Did not touch that code?

rajeshdoshi, 2020-06-16:
Ok. Let me try earlier lib and then latest one and send you video again later in the day. 

One problem I am facing here and it is only with this particular model of module. If I power on module and ESP together, then ESP does not start. But if I remove power to the module and start ESP first and then power on module, it works. 
I tried with higher rating of power supply. Tried adding decoupling capacitors. Can it be that this module has OE inverted and on power on for a moment all the LEDs try to draw a huge current not letting the ESP on? I could not check this on scope. If you can think of any reason? 
I had faced this problem with P10 single colour modules in the past where it used to draw huge current on power on. May be I should try inverting OE.

rajeshdoshi, 2020-06-16:
I checked. With PxMatrix.h from https://github.com/2dom/PxMatrix/blob/d120e24fd9fdc6cad7c84c8f89084ff9a94e0972/PxMatrix.h 
The result is as below

[image]

With Latest PxMatrix.h
https://github.com/2dom/PxMatrix/files/4774028/vid_1.zip

Is there something that I might be doing wrong here?

rajeshdoshi, 2020-06-16:
I am just copying PxMatrix.h file. Not all the files.

rajeshdoshi, 2020-06-18:
I deleted the installed library and reinstalled latest ZIP library.
However, still the result is same.  I have attached video. 
Can you help resolve this issue? 

[m1.zip](https://github.com/2dom/PxMatrix/files/4798015/m1.zip)

All rows are not filled up and the sequence is not correct. Also some dots are being filled twice. Thanks

rajeshdoshi, 2020-06-18:
I compared latest PxMatrix.h and PxMatrix.h in commit
//github.com/2dom/PxMatrix/blob/d120e24fd9fdc6cad7c84c8f89084ff9a94e0972/PxMatrix.h

There is a difference in the ZZIAGG pattern code in both the files.

rajeshdoshi, 2020-06-18:
@2dom I compared latest PxMatrix.h and PxMatrix.h in commit
//github.com/2dom/PxMatrix/blob/d120e24fd9fdc6cad7c84c8f89084ff9a94e0972/PxMatrix.h

There is a difference in the ZZIAGG pattern code in both the files.