2dom/PxMatrix #219: Another new panel?

iMartyn, 2020-06-12:
So I got this panel : https://www.aliexpress.com/item/1000001269409.html?spm=a2g0s.9042311.0.0.5ed64c4diIrYC3

And wired it up according to mostly guesswork ( I assumed the same as the other hub75 boards, but there's no marking on theboard at all).

I used the examples/pattern_test.ino and ran through all the variations and the one that got a clearest actual top line was :
```
  display.begin(16);
  display.setMuxPattern(SHIFTREG_ABC_BIN_DE );
  display.flushDisplay();
```

Yep, I also tried different scan patterns, most made no output at all or worse.

Using master pxMatrix, Wemos D1 (esp8266 smaller but same pinout as the nodemcu).

https://videos.martyn.berlin/videos/watch/dd305cac-43b8-404a-9201-75024055a66d

It seems as if there's a leakage over to the next 7 rows as it goes.

Perhaps I just screwed up the wiring somehow but it feels like I've got it right or would get less similar to intended imagery if anything at all.

xsrf, 2020-06-12:
video doesn't work for me... black screen only

rajeshdoshi, 2020-06-12:
Hmm. It doesnt fill up all the rows. 
Video attached. 

[vid_1.zip](https://github.com/2dom/PxMatrix/files/4771901/vid_1.zip)

iMartyn, 2020-06-12:
sometimes peertube sets the quality as 0p and gives a black screen. The download link is available on the page as well.

xsrf, 2020-06-13:
okay... found the download button. Can you please make sure that your connections are all good? this random flickering suggests really bad connections. What kind of power supply are you using? What is its actual voltage and current rating?

iMartyn, 2020-06-13:
Yep, was indeed a connection issue, not power, but grounding to the micro mostly (noise on other lines too, but for reference, it's now mostly working with :
```
  display.begin(8);
  display.setMuxPattern(SHIFTREG_ABC_BIN_DE);
```
No particular `setScanPattern` - it's still noisy so I'm gonna try soldering the connections rather than relying on dupont connections at the micro end.

closed 2020-06-13 by iMartyn