2dom/PxMatrix #220: AB panel chaining

Josiko85, 2020-06-12:
Good morning, I'll give you my video and the code I use. 
Greetings
`#define PxMATRIX_COLOR_DEPTH 1
#define PxMATRIX_MAX_HEIGHT 16
#define PxMATRIX_MAX_WIDTH 64

#include <PxMatrix.h>
#include <avr/pgmspace.h>
// Pins for LED MATRIX
#define P_A 2
#define P_B 3
#define P_C 0
#define P_D 0
#define P_E 0
#define P_LAT 7
#define P_OE 8

PxMATRIX display(64, 16, P_LAT, P_OE, P_A, P_B);

// Some standard colors
uint16_t myCYAN = display.color565(0, 255, 255);
uint16_t myMAGENTA = display.color565(255, 0, 255);

unsigned long start_time=0;
uint8_t icon_index=0;

void setup() {
     Serial.begin(9600);
     display.begin(2);
     display.setMuxPattern(BINARY);
     display.setScanPattern(ZAGGIZ);
//     display.setBlockPattern(DBCA);
     display.setPanelsWidth(2);
//

display.clearDisplay();
for (int y=0; y<16; y++)
{
    for (int x=0; x<64; x++)
    {
      display.drawPixel(x, y, myCYAN);
      start_time = millis(); 
      while((millis()-start_time)<80)
          display.display(20);
    }
}

     display.clearDisplay();
    display.setTextColor(myCYAN);
     display.setCursor(2,0);
     display.print("1  2  3  4");
     display.setTextColor(myMAGENTA);
     display.setCursor(2,8);
     display.print("5  6  7  8");
     start_time = millis();
     while((millis()-start_time)<300000000)
         display.display(20);
}
void loop() { }`

and one video.
https://youtu.be/WLUwMP4G6VA

xsrf, 2020-06-12:
Can you please confirm that this code draws a straight diagonal white line on both panels when they are connected as single panel each? Please make sure this works on both, so both are actually the same.

```
#define PxMATRIX_COLOR_DEPTH 1
#include <PxMatrix.h>
#include <avr/pgmspace.h>
// Pins for LED MATRIX
#define P_A 2
#define P_B 3
#define P_LAT 7
#define P_OE 8

PxMATRIX display(32, 16, P_LAT, P_OE, P_A, P_B);

void setup() {
    display.begin(2);
    display.setMuxPattern(BINARY);
    display.setScanPattern(ZAGGIZ);
    display.clearDisplay();
    display.drawLine(0, 0, display.width()-1, display.height()-1, 0xFFFF);
}

void loop() {
    display.display(20);
}
```

Btw: To insert code, please paste your code, select it, then click "insert code" button.

Josiko85, 2020-06-13:
> Can you please confirm that this code draws a straight diagonal white line on both panels when they are connected as single panel each? Please make sure this works on both, so both are actually the [...]

Hi, my panels draw a white diagonal, but one of the panels has died because of a connection failure on my part.

In a couple of days I'll have another one and we can continue with the tests.

https://youtu.be/QpGDQ4eOjqA

xsrf, 2020-06-13:
Oh, sorry to hear that :/

Josiko85, 2020-06-13:
> Oh, sorry to hear that

You can't do too many things at once. Talking to people and connecting things isn't usually a good idea....

Josiko85, 2020-06-14:
I'm ready again to continue the tests and adjustments when you can.

xsrf, 2020-06-14:
so, what do you now get with both connected together and this code?

```
#define PxMATRIX_COLOR_DEPTH 1
#include <PxMatrix.h>
#include <avr/pgmspace.h>
// Pins for LED MATRIX
#define P_A 2
#define P_B 3
#define P_LAT 7
#define P_OE 8

PxMATRIX display(64, 16, P_LAT, P_OE, P_A, P_B);

void setup() {
    display.begin(2);
    display.setMuxPattern(BINARY);
    display.setScanPattern(ZAGGIZ);
    display.setPanelsWidth(2);
    display.clearDisplay();
    display.drawLine(0, 0, display.width()-1, display.height()-1, 0xFFFF);
}

void loop() {
    display.display(20);
}
```

Josiko85, 2020-06-14:
@xsrf  In the video you can see the result but I did not intend to give work on the weekend. Thanks for the time dedicated
https://youtu.be/RUy9zXyqA1o

Josiko85, 2020-06-16:
@xsrf  @2dom 
Please don't forget me, see when you can use several panels. I already have everything you need to prove. 
It would be great to be able to use them in chains to even rotate the letters.

xsrf, 2020-06-17:
Please post a video of this with both panels chained. I have no clue what's going on there...

```
#define PxMATRIX_COLOR_DEPTH 1
#include <PxMatrix.h>
#include <avr/pgmspace.h>
// Pins for LED MATRIX
#define P_A 2
#define P_B 3
#define P_LAT 7
#define P_OE 8

PxMATRIX display(64, 16, P_LAT, P_OE, P_A, P_B);

void setup() {
    display.begin(2);
    display.setMuxPattern(BINARY);
    display.setScanPattern(ZAGGIZ);
    display.setPanelsWidth(2);
    display.clearDisplay();
    display.drawLine(0, 0, display.width()-1, display.height()-1, 0xFFFF);
}

void loop() {
    display.displayTestPixel(20);
}
```

Do you have an ESP8266 or ESP32 to compare with?

Josiko85, 2020-06-17:
@xsrf Good afternoon, the problem is that are some new scand 1/2 panels, in another post I spoke with 2dom and he asked me to open a new problem, he changed things in the library to work with only one panel.

I'm using Arduino Mega 2560. 
The video is the same as the video from: [https://youtu.be/RUy9zXyqA1o]

2dom, 2020-06-19:
The curious thing is that a single  panel worked just fine with the settings before ? Can you try each panel you intend to chain together individually (as single panels) to make sure they are identical?

Josiko85, 2020-06-19:
@2dom Good afternoon, I made some tests with separate panels and panels together, here I put 5 videos for you to see. I remain at your disposal.

[](https://youtu.be/zrNwqhuzu7s)    panel type
[](https://youtu.be/DDIGeoH8Ixs)     panel A
[](https://youtu.be/KuQkobdrV6E)    panel B
[](https://youtu.be/4MowCbVrCBA)  2-panel
[](https://youtu.be/yR9c1uTkCrcl)     scroll with 2 panel

xsrf, 2020-06-20:
Okay, so both panels act as I would expect it - on their own and also when chained together.
The problem is with the `ZAGGIZ` implementation for more than one panel.

Josiko85, 2020-06-20:
> Okay, so both panels act as I would expect it - on their own and also when chained together.
> The problem is with the `ZAGGIZ` implementation for more than one panel.

If you need me to do any more tests, tell me and I'll send you the life or anything.

2dom, 2020-06-21:
I will have a look as soon as I can ... Very busy ATM

Josiko85, 2020-06-21:
> 
> 
> I will have a look as soon as I can ... Very busy ATM
You don't have to worry. Whenever you can, thank you.

Josiko85, 2020-07-03:
@2dom 
Hello, when you can remember me. If you can get three cascading panels to work, you've got another virtual beer guaranteed for the effort.

And one more question, do I want to put "Val" but if I put the size at 2 it doesn't catch me, can I change the font or something?

Thank you.