2dom/PxMatrix #224: P10 Outdoor Panel weird Test Pattern

BusbuyukHalter, 2020-06-20:
With the provided pattern test sketch results are like in the video. I tried using STRAIGHT mux pattern and all of the scan patterns but results are pretty much same.

[https://www.youtube.com/watch?v=JcZmrQnkf3U&feature=youtu.be](url)

```
#include <PxMatrix.h>

#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#endif
// Pins for LED MATRIX

PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);
//PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
//PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

#ifdef ESP8266
// ISR for display refresh
void display_updater()
{
  display.displayTestPattern(70);
  // display.displayTestPixel(70);
  //display.display(70);
}
#endif

#ifdef ESP32
void IRAM_ATTR display_updater(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  //isplay.display(70);
  display.displayTestPattern(70);
  portEXIT_CRITICAL_ISR(&timerMux);
}
#endif

uint16_t myCYAN = display.color565(0, 255, 255);
void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
  display.begin(4);
  display.flushDisplay();
  display.setTextColor(myCYAN);
  display.setCursor(2,0);
  display.print("Pixel");
  Serial.println("hello");

  #ifdef ESP8266
    display_ticker.attach(0.004, display_updater);
  #endif

  #ifdef ESP32
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &display_updater, true);
    timerAlarmWrite(timer, 4000, true);
    timerAlarmEnable(timer);
  #endif

  delay(1000);
}

void loop() {

 delay(100);

}
```

BusbuyukHalter, 2020-06-20:
This is the panel I'm using.
[image]
[image]
[image]

xsrf, 2020-06-20:
You said you tried `STRAIGHT` but the code uses `BINARY`.

Whatever was used in the Video is correct.

If it was `STRAIGHT` you need to connect and define Pins A-D.

If it was `BINARY` you need to connect and define Pins A-B.

To check for the right scan pattern use an example that actually shows an image/text please or just draw a diagonal line. The scan-pattern has no effect on `displayTestPattern` or `displayTestPixel`

hsynklceng, 2022-04-06:
Hi, did you solved your problem?