2dom/PxMatrix #226: Red channel shifting right by 1 (in one area)

CameronH1993, 2020-06-21:
```
// This is how many color levels the display shows - the more the slower the update
//#define PxMATRIX_COLOR_DEPTH 4

// Defines the speed of the SPI bus (reducing this may help if you experience noisy images)
//#define PxMATRIX_SPI_FREQUENCY 20000000

// Creates a second buffer for backround drawing (doubles the required RAM)
#define PxMATRIX_double_buffer true

#include <PxMatrix.h>

// Pins for LED MATRIX
#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 16
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#endif

#define matrix_width 128
#define matrix_height 32

#define RGB 565

#if RGB==565
  #define frame_size matrix_width*matrix_height*2
#else
  #define frame_size matrix_width*matrix_height*3
#endif

uint16_t frame_no=0;
unsigned long anim_offset=0;

union single_double{
  uint8_t two[2];
  uint16_t one;
} this_single_double;

// This defines the 'on' time of the display is us. The larger this number,
// the brighter the display. If too large the ESP will crash
uint8_t display_draw_time=30; //30-70 is usually fine

//PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);
//PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
PxMATRIX display(matrix_width,matrix_height,P_LAT, P_OE,P_A,P_B,P_C,P_D);

// Some standard colors
uint16_t myRED = display.color565(255, 0, 0);
uint16_t myGREEN = display.color565(0, 255, 0);
uint16_t myBLUE = display.color565(0, 0, 255);
uint16_t myWHITE = display.color565(255, 255, 255);
uint16_t myYELLOW = display.color565(255, 255, 0);
uint16_t myCYAN = display.color565(0, 255, 255);
uint16_t myMAGENTA = display.color565(255, 0, 255);
uint16_t myBLACK = display.color565(0, 0, 0);
uint16_t my_color = display.color565(255, 32, 0); //g was 32
uint16_t myCOLORS[8]={myRED,myGREEN,myBLUE,myWHITE,myYELLOW,myCYAN,myMAGENTA,myBLACK};

#ifdef ESP8266
// ISR for display refresh
void display_updater()
{
  display.display(display_draw_time);
}
#endif

#ifdef ESP32
void IRAM_ATTR display_updater(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  display.display(display_draw_time);
  portEXIT_CRITICAL_ISR(&timerMux);
}
#endif

void display_update_enable(bool is_enable)
{

#ifdef ESP8266
  if (is_enable)
    display_ticker.attach(0.001, display_updater);
  else
    display_ticker.detach();
#endif

#ifdef ESP32
  if (is_enable)
  {
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &display_updater, true);
    timerAlarmWrite(timer, 2000, true);
    timerAlarmEnable(timer);
  }
  else
  {
    timerDetachInterrupt(timer);
    timerAlarmDisable(timer);
  }
#endif
}
void setup() {

 Serial.begin(9600);
  // Define your display layout here, e.g. 1/8 step, and optional SPI pins begin(row_pattern, CLK, MOSI, MISO, SS)
  display.begin(16);
  //display.begin(8, 14, 13, 12, 4);
  display.setPanelsWidth(2);
  // Set driver chip type
  display.setDriverChip(FM6124);
  //display.setBrightness(254);
  display.clearDisplay();
  display.setTextColor(my_color);
  display.setCursor(1,0);
  display.print("000000000000000000000");
  
  display.setCursor(1,8);
  display.print("000000000000000000000");

  display.setCursor(1,16);
  display.print("000000000000000000000");

  display.setCursor(1,24);
  display.print("000000000000000000000");

  display_update_enable(true);
  display.showBuffer();
  

  delay(1000);
  
    Serial.println("Initi complete");

}

void loop() {

 delay(100);

}
```
[image]

I can't figure out why there is a shifting on the red channel as seen above. I have tried changing speed (SPI/MUX), I have tried fastmode and rechecked every connection. Any ideas?!

TEST PATTERN:
[image]

I'm using WEMOS D1 (ESP8266) and PxMatrix v1.8.2

Any help appreciated!

xsrf, 2020-06-21:
`display.setDriverChip(FM6124);` did you verify you actually need this?

CameronH1993, 2020-06-21:
I feel stupid now but I'm glad that fixed it! haha

Cheers!

closed 2020-06-22 by CameronH1993