2dom/PxMatrix #227: Some parts of the display disappear from time to time

lg007, 2020-06-28:
First of all, I congratulate for this fantastic library.

I am running the latest library with an ESP32 and a GPRS modem to display numbers on a sinlge 64x32 RGB P5 matrix using MQTT. Everything works right, but sometimes part of the display disappears and then - when the shift register pumps in data - appears again. I double checked the code, it happens very randomly and apparently it is not connected with the GPRS functionality. If it helps I also have - but very rarely - some flickers as well.

2dom, 2020-06-28:
Hi...the shifting out data happens on a ms scale so you would only be able to perceive this as flicker. If the display visibly goes blank it is usually to some code blocking the CPU. To test this simply run the PixelTime example. If everything works there the observed behaviour is probably due to you other code

lg007, 2020-06-28:
Thanks for the prompt answer. I was thinking about the same thing, so you hit the problem. Thanks anyway, I will try to find what is blocking the the CPU, but I have the feeling it is the GPRS / MQTT library.

closed 2020-06-28 by lg007