2dom/PxMatrix #229: ESP32, PxMatrix and OTA not working for me.

richms, 2020-07-04:
I thought I would have a shot with arduino OTA uploading on my wemos 32 board in Brian Lough's adapter board.

Its all working fine with the basic PxMatrix pixel time sketch configured for my 3 panels that are connected.

If I put a while loop on the start of setup, delaying the startup of the sketch it all works fine to upload to it.

`void setup() {

  Serial.begin(115200);
  Serial.println("Booting");
 
  setupOTA("scrollingsign");

  Serial.println("OTA waiting");
  long start = millis();  // wait for 10 second to process OTA stuff.
  while ( millis()-start <= 30000 ) {
    ArduinoOTA.handle();
  } 
  Serial.println("No OTA, continueing");

  // Define your display layout here, e.g. 1/8 step, and optional SPI pins begin(row_pattern, CLK, MOSI, MISO, SS)
  display.begin(8);

Etc etc etc going on .
`

All works fine there, but I have a delay on startup. At least I can change the text by starting an upload and going an hitting reset on the wemos module.

But if I put the     ArduinoOTA.handle(); any point after starting the display up - even in a while loop in setup() I get 

`void setup() {

  Serial.begin(115200);
  Serial.println("Booting");
 
  setupOTA("scrollingsign");

//  Serial.println("OTA waiting");
//  long start = millis();  // wait for 10 second to process OTA stuff.
//  while ( millis()-start <= 30000 ) {
//    ArduinoOTA.handle();
//  } 
//  Serial.println("No OTA, continueing");

  // Define your display layout here, e.g. 1/8 step, and optional SPI pins begin(row_pattern, CLK, MOSI, MISO, SS)
  display.begin(8);
  //display.begin(8, 14, 13, 12, 4);

  // Define multiplex implemention here {BINARY, STRAIGHT} (default is BINARY)
  //display.setMuxPattern(BINARY);

  // Set the multiplex pattern {LINE, ZIGZAG,ZZAGG, ZAGGIZ, WZAGZIG, VZAG, ZAGZIG} (default is LINE)
  //display.setScanPattern(LINE);

  // Rotate display
  //display.setRotate(true);

  // Flip display
  //display.setFlip(true);

  // Helps to reduce display update latency on larger displays
  //display.setFastUpdate(true);

  // Control the minimum color values that result in an active pixel
  //display.setColorOffset(5, 5,5);

  // Set the multiplex implemention {BINARY, STRAIGHT} (default is BINARY)
  //display.setMuxPattern(BINARY);

  // Set the color order {RRGGBB, RRBBGG, GGRRBB, GGBBRR, BBRRGG, BBGGRR} (default is RRGGBB)
  //display.setColorOrder(RRGGBB);

  // Set the time in microseconds that we pause after selecting each mux channel
  // (May help if some rows are missing / the mux chip is too slow)
  //display.setMuxDelay(0,1,0,0,0);

  // Set the number of panels that make up the display area width (default is 1)
  display.setPanelsWidth(3);

  // Set the brightness of the panels (default is 255)
  display.setBrightness(50);

  // Set driver chip type
  //display.setDriverChip(FM6124);

  display.setFastUpdate(true);
  display.clearDisplay();
  display.setTextColor(myCYAN);
  display.setCursor(2,0);
  display.print("Pixel");
  display.setTextColor(myMAGENTA);
  display.setCursor(2,8);
  display.print("OTA 60sec");
  display_update_enable(true);

  Serial.println("OTA waiting");
  long start = millis();  // wait for 60 second to process OTA stuff.
  while ( millis()-start <= 60000 ) {
    ArduinoOTA.handle();
  } 
  Serial.println("No OTA, continueing");

  display.setCursor(2,8);
  display.print("OTA Done");
}`

`Booting
OTA Initialized
IP address: 172.16.38.86
OTA waiting
Start updating sketch
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149e38: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149e38: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149e38: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
`
while it tries to upload to it.

Has anyone got the OTA uploading and PxMatrix working together on an ESP32 board?

richms, 2020-07-04:
I dont know why its not taking code tags when I click it Sorry for the bad formatting.

xsrf, 2020-07-04:
The display routine blocks the CPU for a long time, making huge network transfers unreliable. I'd recommend to disable the display during update via the onStart() callback of the OTA library or at least drastically reduce the timer interval.

richms, 2020-07-05:
I thought it was stopping that already since the led panel blanks out when it tries to start the flash. I will have to have another play with it at a later stage because it is up and working now with a long USB cable to let me reflash with alternate messages. I need to sort out something to change it over the air and just reflashing was the easiest option I thought of at the time.

txf-, 2020-07-09:
> The display routine blocks the CPU for a long time, making huge network transfers unreliable. I'd recommend to disable the display during update via the onStart() callback of the OTA library or at least drastically reduce the timer interval.

About this, presumably one could run the OTA process on a different core?

xsrf, 2020-07-09:
Sure... But I don't see why one would put much effort into that. OTA is usually part of a maintenance and it shouldn't be a problem to turn the display off to do that. It's an easy solution.

On the other hand, if you have huge network transfer as part of the regular operation, you should consider optimizing the code this way.

txf-, 2020-07-09:
One might be using the the same way one would a screen, possibly to show a progress bar or an icon to indicate that the device is updating.

hallard, 2020-07-26:
The deal is that it's not related to OTA only see #113, I've got a web server also and each time I go on it to do setup config it just crash. And it works fine on ESP8266 with
- pxMatrix
- ESPAsyncWebServer
- AsyncMqttClient
- of course SPIFFS (LittleFS on esp8266)
- OTA with progress bar display

But I need other GPIO so that's why I use now ESP32 not using any func using SPIFFS (WebServer, OTA).

Not sure how we can handle that but for sure using only 2nd core on ESP32 to manage display would be awesome

chartinger, 2020-08-03:
It seems updating the display from within the ISR will end up in a world of hurt on the ESP32. From i2c infinite loops (caused by https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-i2c.c#L1010) and `xQueueGenericReceive` errors from SPI it is a debug hell.

What I did on the esp32 is delegating the display work to a task (pinned to core 1) and task-notify it from the ISR. The Task hast to be priority > 3 as the AsyncWebServer has 3 as a hardcoded magic priority and values equal or below will flicker the matrix every time there is a request. While OTA+Matrix technically works, it flickers the whole duration. (Strangely, trying to pin OTA to core 0 resulted in even more flickering)

txf-, 2020-08-03:
> It seems updating the display from within the ISR will end up in a world of hurt on the ESP32. From i2c infinite loops (caused by https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-i2c.c#L1010) to `xQueueGenericReceive` errors from SPI it is a debug hell.
> 
yes, I encountered this too. I documented my solution here #225 , whereby the interrupt routine actually forces the scheduler to call the thread handling the screen update. 

Regarding the flickering, I wonder if it has anything to do with reading and writing to flash while attempting to use the matrix. IIRC this is also a reported issue, though that was regarding SPIFFS.