2dom/PxMatrix #231: Trying to figure out settings for a generic chinese P5 64x32 matrix with seemingly unsupported pattern

BodoMinea, 2020-07-13:
_1. Post your code_ 
**I do not think the code is the root of the problem here, because the same code with the same microcontrollers works with a different type of panel (P4 64x32 works with the default settings) - https://www.aliexpress.com/item/4000045356194.html?spm=a2g0s.9042311.0.0.180a4c4dA8MLig**
```
#define matrix_width 64
#define matrix_height 32

#include <PxMatrix.h>

#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 16
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#endif

uint8_t display_draw_time=20;

#ifdef ESP8266
// ISR for display refresh
void display_updater()
{
  display.display(display_draw_time);
}
#endif

#ifdef ESP32
void IRAM_ATTR display_updater(){
  portENTER_CRITICAL_ISR(&timerMux);
  display.display(display_draw_time);
  portEXIT_CRITICAL_ISR(&timerMux);
}
#endif

void display_update_enable(bool is_enable){
  #ifdef ESP8266
    if (is_enable)
      display_ticker.attach(0.002, display_updater);
    else
      display_ticker.detach();
  #endif
  
  #ifdef ESP32
    if (is_enable)
    {
      timer = timerBegin(0, 80, true);
      timerAttachInterrupt(timer, &display_updater, true);
      timerAlarmWrite(timer, 2000, true);
      timerAlarmEnable(timer);
    }
    else
    {
      timerDetachInterrupt(timer);
      timerAlarmDisable(timer);
    }
  #endif
}

void setup() {
  display.begin(16);
  display.setFastUpdate(true);
  display.clearDisplay();
  display.setCursor(2,0);
  display.print("Pixel");
  display.setCursor(2,8);
  display.print("Time");
  display_update_enable(true);
}

void loop() {

}
```
_2. Post a picture of the problem and describe what you expect to see_
**In these pictures I will whow you what comes up with different settings while trying to display the text „1B” with a color close to red (they are in somewhat of an order, from best to worst outcomes):**
**BINARY mux - WZAGZIG2 scan**
[image]
**BINARY mux - WZAGZIG scan**
[image]
**BINARY mux - LINE scan (lib default)**
[image]
**BINARY mux - VZAG scan**
[image]
**STRAIGHT mux - LINE scan (seems to mess it up even further so I do not think changing the mux from BINARY would do any good)**
[image]
**It is my understanding that this is an 1/16 panel. I have also tried 1/8 scan with the other settings on default and the result was botched.**
_3. Run the pattern test and post the result as gif/video (need to have a bit more than one full RED/YELLOW/WHITE cycle)_
**I am attaching you videos of the standard tests.**
[Pixel pattern test](https://youtu.be/kerpI4K2cYc)
[Lines pattern test](https://youtu.be/Jt3_-BPB4pA)
_4. State what version of PxMatrix you are running and what MicroController you use_
**I have tried with the v1.8.2 May release and also the latest master zip from here. The results are the same. All the tests above were tried on a Lolin ESP8266 and an ESP32 DevKit with the same results.**
I am currently trying to obtain more information in regards to this panel's specs, but if someone can suggest something to try in the meantime, I will be glad.

I thank you in advance for your time and for keeping up this great project!

xsrf, 2020-07-14:
What settings were you running the pattern_test on? I guess begin(16)? Your panel looks like 1/8. Can you run the pattern test again with begin(8) please?

BodoMinea, 2020-07-18:
You were right in thee fact that it is 1/8, a fact that was also confirmed by the manufacturer. The colors now look right and the pattern_test now looks correct. However, I still can't seem to find the correct setting.

Wzagzig - displaying „1B” text centered
[image]
Wzagzig - displaying „3B” text centered and a line between the two corners (correctly)
[image]
Wzagzig - displaying „9” text centered and a line between the two corners (correctly), inverted
[image]
Vzag - displaying „1B” text centered
[image]
Line - displaying „3B” text centered
[image]
Line - displaying „1B” text centered
[image]
Wzagzig2 - displaying „1B” text centered - the closest to correct I got. The „1B” should've been displayed together.
```
display.setScanPattern(WZAGZIG2);
display.begin(8);
display.setTextColor(customColor);
display.setCursor(14,1);
display.setTextSize(3);
display.print("1B");
```
[image]

Sorry for the late reply but I do not have the panel at my house but rather at the office where I am not coming by daily.

BodoMinea, 2020-07-22:
Solved. This panel is internally composed of two separate 32x32 matrices and using the WZAGZIG2 scan pattern. 
So it is working as intended with
```
PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
display.begin(8);
display.setPanelsWidth(2);
display.setScanPattern(WZAGZIG2);
```

closed 2020-07-22 by BodoMinea