2dom/PxMatrix #233: One 32x64 works, great,  but three in a row daisy chained to make 32x196  doesnt work

jerkwagon, 2020-07-16:
## Rules
1. Post your code (i will edit this shortyly
This is what im trying to do https://i.imgur.com/i3ThDw8.jpg

3. Works fine a single 32x64
 
https://imgur.com/6ddeNsr

heres my 1x3 array 

https://imgur.com/JAVux1f

4. Latest version of pxmatrix and esp32s

okelk, 2020-10-13:
I'm also trying to chain 32x64pixel matrices (p5 in my case). I'm using an esp8266.
chaining two works just fine.
with 3 I get strange artefacts.

jerkwagon, 2020-10-13:
> I'm also trying to chain 32x64pixel matrices (p5 in my case). I'm using an esp8266.
> chaining two works just fine.
> with 3 I get strange artefacts.

i ended up using a pi instead, 100x easier.