2dom/PxMatrix #236: PIN defintion in examples for ESP32 are wrong

feiticeir0, 2020-08-13:
Fist, thanks for the library  ! Works wonderfuly ! 

I'm using version 1.8.1

In the examples, the PINs definition for ESP32 have a bug for P_OE .
The PIn definition is 2 (the same for ESP8266) but in the wiring schematics in README.MD is 16 .

Just a heads up. I've spent two nights wondering why it works on ESP8266 and not on ESP32 :) 
Until I look better to the code and spot the culprit. 

`// Pins for LED MATRIX
#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2  **<- Should be 16**
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif`

Best regards

kevensmiley, 2020-08-20:
I find this to be incorrect OE= 2

catSIXe, 2020-08-20:
Yes, all my struggle getting it flicker free.... It was prob. triggering by the data bus interference instead. So this fixed all issues with ghosting for me.

catSIXe, 2020-08-20:
```

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 0
#define P_OE 2
PxMATRIX display(64, 32, P_LAT, P_OE, P_A, P_B, P_C, P_D, P_E);
```

works for my 64x32 P4 Display on ESP32

Dutchy-, 2021-01-01:
I had the same problem, the symptom was a too bright line 16 and 32 on a P6 32x32 board, with the rest of the board working fine. Setting P_OE 16 fixed the problem! Thanks @feiticeir0 !

2dom, 2021-01-01:
OK...Cheers. I moved ist from 2 to 16 as 2 is a strapping pin that may interfere with downloading code to the esp. must have missed to update  the example

fruktiz32, 2021-12-04:
Thanks, it worked for me!