2dom/PxMatrix #240: Printing characters result in  inverse symetrics over Y axis shown.

Antoniohopu, 2020-09-01:
## Rules
1. Post your code 
2. Post a picture of the problem and describe what you expect to see
3. Run the pattern test and post the result as gif/video (need to have a bit more than one full RED/YELLOW/WHITE cycle)
4. State what version of PxMatrix you are running and what MicroController you use
  
Hi, I'm using this library with a 32x16 Chinese panel with 1/4 scan and the patter_test.ino works well but when I execute or run the pixeltime.ino projects, all the characters are shown ina inverse symmetric way over Y rotation axis. 
I mean when I triy to draw the character 'e'  it is shown a '9' in my panel.
I have been testing different configurations and functions but without any valid results. I'm using a esp32 wroober-b and the last one available from github:https://github.com/2dom/PxMatrix.

[image]

Could you help me or at least, give me some recommendations in order to fix this problem.. I also attach you an example where the character 'e' should be shown but it is shown them'9' duo to the image effect over the Y axis.

xsrf, 2020-09-01:
you have to change the SPI bit order. Try `SPI.setBitOrder(LSBFIRST)` or `SPI.setBitOrder(MSBFIRST)` after initializing the panel.

Antoniohopu, 2020-09-08:
Hi,
the problem was that the flip option was enabled.  With   display.setFlip(false) the problem was resolved.