2dom/PxMatrix #243: P10 16x32 2 chained panels vertically

federicoprimomo, 2020-09-24:
Hi, I'm trying to chain vertically 2 p10 16x32 panels.

Inicialization 

`  display.begin(4);
  display.setPanelsWidth(2);
  display.setScanPattern(ZAGGIZ);
  display.setMuxPattern(BINARY);
  display.setFastUpdate(true);
  display.clearDisplay();
  display_update_enable(true);`

For example, writing a rectangle on the corners works great:

  display.fillRect(0, 0, 5, 5, myGREEN);
  display.fillRect(26, 0, 31, 5, myWHITE);
  display.fillRect(0, 26, 5, 31, myBLUE);
  display.fillRect(26, 26, 31, 31, myRED);

The problem only it's on the second half of the 1st panel and the 1 half of the 2 panel.

Any ideas ?

federicoprimomo, 2020-09-24:
See this code, and how it's displayed:
`  int c=0;
  while(true){
    while(c<=32){
      display.fillRect(0, 0, c, c, myYELLOW);
      c++;
      delay(100);
    }
    c=0;
    while(c<=32){
      display.fillRect(0, 0, c, c, myWHITE);
      c++;
      delay(100);
    }
    c=0;
    while(c<=32){
      display.fillRect(0, 0, c, c, myRED);
      c++;
      delay(100);
    }
  }`

[image]

rajeshdoshi, 2021-09-06:
I have same problem. Were you able to resolve this issue?