2dom/PxMatrix #247: 64x64 panel blue artifacts and weird colours

CoreyInDasHaus, 2020-10-10:
Git hub V1.8.2 fc53edf

64x64 SRYLED  P2.5-2121-32S-JJ-V6 bought two panels from affiliate link from Youtuber Bitluni: http://s.click.aliexpress.com/e/bhQPhfeY

Running on D1 mini
Wiring diagram from Witnessmenow

Chipset: Same as Maditnerd issue (119) - they had 32x64 panel

	•	RuiChips RUC7258 (Led Gate Driver) : http://www.ruichips.com/uploads/file/20180816/20180816053859223.pdf
	•	Chipone ICN2037 (16-Channel Constant Current LED Sink Driver with Dual Latch) http://en.chiponeic.com/content/details36_245.html
	•	Chipone ICN74HC245 (Logic Control) http://en.chiponeic.com/content/details38_95.html

In order to check my wiring, I used set up two D1 mini and two panels powered by a computer power supply:
pic of side by side.
[image]

For my sketches, I use:
PxMATRIX display(64, 64, P_LAT, P_OE, P_A, P_B, P_C, P_D, P_E); **P_E is not grounded**
display.begin(32);

Out of the box, the Mario sketch is like so:
[image]
but adding display.setMuxDelay(1,1,0,0,0), a much better image (shown in mario side by side and below), but with a weird blue bar and a green sky and cyan Earth. I also have a blue bar at the bottom of my screen. code [mario.txt](https://github.com/2dom/PxMatrix/files/5359781/mario.txt)
[image]

Ive tried all combinations of muxdelay from 0,0,0,0,0 to 1,1,1,1,1.

When I run the test.ino with PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E),   display.begin(32);
I get this (.gif sped up 3x) : code [pattern test.txt](https://github.com/2dom/PxMatrix/files/5359920/pattern.test.txt)

[image]

I get red, yellow yellow, but no white!

note that if I do not run display.setMuxDelay(1,1,0,0,0), after the fourth cycle of testing, the row becomes very erratic flickering on and off then the next line is ok, but still no white.

As I was typing this issue out I left the test running, on the 17th iteration (for both no mux delay and mux display) a blue line appeared first and then went magenta, white, white:
[image]

Lastly, this is my double buffer test, with the weird blue box artifacts.
[image]

My goal is to get the aurora demo working, but that seems to not like when there is a mux delay. 

Any help on this would be awesome! Thanks so much!

syed93uthman, 2020-12-25:
I also have this issue. Looking forward for a solution

Hcreak, 2023-03-28:
LOL. It may be because the color mode of the upper and lower fields is different. I have this problem with the LED panel in my hand.