2dom/PxMatrix #248: Anyone else having issues with wrong colors?

johniwad, 2020-10-23:
I'm using a generic D1 mini along with a D1 Mini Matrix Breakout v1.4.0 and P3 64x32 LED matrix I purchase off of ebay. Everything seems to work OK but the colors are all wrong. Does anyone know where or how to make changes so colors come out correct?
[image]
[image]
[image]

2dom, 2020-10-23:
SetColorOrder could help

davemaster, 2020-10-23:
greetings

Apparently you miss some clables pin location, check the RGBR1G1B1 cables
again... one is in miss-order

Best regards

johniwad, 2020-10-23:
I was afraid of pin connectors in wrong order but it is a breakout board so it wasn't my wiring. I was thinking there must be a way to change in software but don't know how.