2dom/PxMatrix #249: Can't get any response from display

VictorHofstede, 2020-10-26:
Yesterday I was wiring up a 64x32 P4 display to my ESP32 using dupont cables. I got it to a point where the whole screen was flickering white but that only happened when I moved and held the dupont cables in the breadboard. I gave up for the night and started again this morning with soldering everything in place. I noticed some wrong wiring and thought I had fixed everything.

But since yesterday evening I haven't been able to get anything from my display. Not even one pixel showed signs of life. I measured with a multimeter every wire for continuity at least 3 times and every wire is connected according to the wiring schema. 

What can I do next to debug?

If needed I can include some photos of my board, display and wiring. This is my first GitHub ticket ever so let me know if it requires more info.

VictorHofstede, 2020-10-26:
Here is a general overview of my wiring.
[image]

The matrix I've got is
https://www.aliexpress.com/item/32754106669.html

VictorHofstede, 2020-10-27:
I just checked my wiring again with a multimeter and everything is according to the diagram. The only thing I can think of is that the LEDs are burnt out but I can see no evidence of that. Does anyone know how I can measure whether something in the panel itself is broken?

VictorHofstede, 2020-10-27:
I've now tried a different library (https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA) and that sort of works. I only got the top half of the screen working. For the bottom half only the green pixels work. 

If anyone reads this and has the slightest idea what and why stuff is going wrong, please help me out

xsrf, 2020-10-28:
How do you connect Power to your Microcontroller? Did you connect GND of the panel connector to GND on the Microsocontroller? Can you try to run you Panel with 3,3V to 4V, not 5V?
Also make sure to use cables that have really good connection.
If it works somehow, please post a Video and the exact code used.

VictorHofstede, 2020-10-29:
I use an ESP32 as microcontroller which I power through USB. I connected all GND pins from the panel to the ESP32 GND. I run the panel on 5V but can try going a little lower (eventhough the specs say 5VDC)

For the connection I soldered wires onto pinheaders that I connected to my ESP32 and to the ribbon cables that go into the 16-pin connectors on the panel. I did this to make sure nothing is permanent yet. Do you think this could cause I bad connection somewhere?

VictorHofstede, 2020-10-29:
[image]
Here is a close-up of my soldering. I think it's pretty okay and with a multimeter I measured the continuity between all the pins so I didn't make any mistake in connecting the wrong pins. 

A video won't be of much use I think since I get absolutely no response. No flickering, nothing.

xsrf, 2020-10-29:
The panel is specified for 5V, correct. But you're driving it with an ESP that provides logic levels of only 3.3V or less.
Most logic ICs specify that a `HIGH` signal needs to be at least `0.7 * Vcc`. So when you drive your panel with 5V, a logic level `HIGH` needs to be above 3.5V to be withing specification.
You can either use a 5V Microcontroller, or drive the panel with less voltage to meet the specification.

I'd also suggest to start over with the bare minimum. Disconnect all wires from the output of the panel and just use one color at the input (R0) for debuging. When you get that working, add back the chaining of the other color channels.

VictorHofstede, 2020-10-29:
Thanks for your reply. I did not know the 5V power supply on the board could be a problem. I did however follow a video tutorial from Brian Lough (https://www.youtube.com/watch?v=YvU_ZfF7vs4) and he uses an ESP8266 and a 5V laptop charger as PSU. How come that works for him?

xsrf, 2020-10-29:
Because most 5V logic ICs register 3.3V as HIGH despite not being within spec. Don't get me wrong. 5V supply with 3.3V logic will work on almost all panels. However, for debugging, you should test if that may be the issue.

VictorHofstede, 2020-10-29:
I do not have a power supply that can deliver 3.3-4V reliably. Only a 5V regulated power supply that I can tune down to 4.5V. 
I did try with all the wires from the output disconnected but it gave me the same result.

Yesterday I also tried the library I linked above (https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA) and I did get that to work. It requires all the output wires to be hooked up to the ESP32 aswell, but other than that I do not see something significantly differnet from that library. Except that it looked a little harder to actually work with.

VictorHofstede, 2020-11-08:
I am still stuck on this issue and would love if someone could help me out with more ideas. 

I tried to solder the wires directly to the ESP32 instead of using pinheaders and that gave me some response. I saw some flickering but only if I held my pinheaders (connected to the pins on the display) in a certain way. I figured my soldering was probably not solid enough so desoldered everything and just attached dupont cables. But now I get no response at all... 

It all looks so simple in the wiring schema but it's not working for me and that is very frustrating