2dom/PxMatrix #250: Shifted colors and shifted lower half with multiple panels

merlinschumacher, 2020-10-27:
Hi, 

I have three panels that show shifted colors all over, and a shifted lower half when daisy-chained together. A single panel seems to work fine. Adding a second induces a 1 pixel shift. Adding a third adds another pixel shift.

Here is an example:
[image]

The test patterns work fine. I've uploaded a video of the test pattern, followed by the mario test image here [https://www.youtube.com/watch?v=I0p8bEszOuA](https://www.youtube.com/watch?v=I0p8bEszOuA)

I use an ESP32 with 1/8th and already unsuccessfully played around with mux and scan patterns. Setting a mux delay didn't change anything.

merlinschumacher, 2020-10-27:
This is the code for the Mario image

``` c++
#include "PxMatrix.h"

#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 16
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#endif
// Pins for LED MATRIX

#define matrix_width 96
#define matrix_height 16

// This defines the 'on' time of the display is us. The larger this number,
// the brighter the display. If too large the ESP will crash
uint8_t display_draw_time=20; //10-50 is usually fine

 PxMATRIX display(matrix_width,matrix_height,P_LAT, P_OE,P_A,P_B,P_C);
//PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
//PxMATRIX display(matrix_width,matrix_height,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

// Some standard colors
uint16_t myRED = display.color565(255, 0, 0);
uint16_t myGREEN = display.color565(0, 255, 0);
uint16_t myBLUE = display.color565(0, 0, 255);
uint16_t myWHITE = display.color565(255, 255, 255);
uint16_t myYELLOW = display.color565(255, 255, 0);
uint16_t myCYAN = display.color565(0, 255, 255);
uint16_t myMAGENTA = display.color565(255, 0, 255);
uint16_t myBLACK = display.color565(0, 0, 0);

uint16_t myCOLORS[8]={myRED,myGREEN,myBLUE,myWHITE,myYELLOW,myCYAN,myMAGENTA,myBLACK};

// Converted using the following site: http://www.rinkydinkelectronics.com/t_imageconverter565.php
uint16_t static mario[] = {
  0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF,   // 0x0010 (16) pixels
  0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF,   // 0x0020 (32) pixels
  0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF,   // 0x0030 (48) pixels
  0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF,   // 0x0040 (64) pixels
  0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF,   // 0x0050 (80) pixels
  0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF,   // 0x0060 (96) pixels
  0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF,   // 0x0070 (112) pixels
  0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF,   // 0x0080 (128) pixels
  0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF,   // 0x0090 (144) pixels
  0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF,   // 0x00A0 (160) pixels
  0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF, 0x07FF,   // 0x00B0 (176) pixels
[... 246 lines cut ...]
  else if(cnt == 3){
    // drawImage2(0, 16);
    drawImage(x, y, myGREEN);
    Serial.println("test green");
    // cnt = 0;
  }
  else if(cnt == 4){
    //drawImage2(0, 32);
    drawImage(x, y, myBLUE);
    Serial.println("test blue");
   // cnt = 0;
  }
  else if(cnt == 5){
    drawImage2(0, 0);
    Serial.println("test mario 1");
    // cnt = 0;
  }
  else if(cnt == 6){
    drawImage2(0, 16);
    Serial.println("test mario 2");
    cnt = 0;
  }

}
```

domingopestana, 2022-05-08:
Hi!
I just registered to github just to reply here...

I was having the same problem and then it struck me... what I was seeing was the data buffers "loosing" one clock cycle. Or better, clocking one data cycle late on every new row, which only happens after going through the longer side of the cable.

I slowed the SPI bus clock speed and that solved the issue. To do that, you have to define the following variable:
`#define PxMATRIX_SPI_FREQUENCY 10000000`

BEFORE calling  
`#include <PxMatrix.h>`

The original value for it was 20000000, and I slowed it to half. I believe not such a drastic reduction is needed but this is good for me!

Also, please take care in the examples it is incorect spelled, so please, double check!

A last note, the issue is not with the led drivers, Chipone ICN2037pb in my case, but with the buffers, SM74HC245 in my case.