2dom/PxMatrix #251: ICN2038S panel shifted colours when chained

dom96, 2020-11-01:
So I got two panels with the CHIPONE ICN2038S chip ([icn2038s.pdf](https://github.com/2dom/PxMatrix/files/5470822/icn2038s.pdf)), they're P3 and 64x32. I'm guessing that this chip is not supported?

It seems to be working fine when I use just one panel. But as soon as I chain two I get colour shifting and pixel flickering.

As an example I ran this code: https://gist.github.com/dom96/6d5c142be99118bb42ac6766d2633aef

[image]

It should show a yellow rectangle (the camera ruins the colours, but the middle is yellow, with green on the left and red on the right). Note that the display is upside down, so 0,0 is bottom right.

When connected to just one panel:

[image]

Drawing text results in this sort of mess:

[image]

Whereas if I just connect one panel it looks fine:

[image]

I tried a bunch of different things, I did a continuity test on all my pins (they are all connected where the readme specifies)., I tried setting mux delays, I tried pretty much all the mux patterns/chips/etc. Nothing really helps. So my assumption is that the chip is to blame.

merlinschumacher, 2020-11-02:
I have a very similar issue in #250

preet808, 2020-11-04:
https://github.com/2dom/PxMatrix/issues/252#issue-735980802