2dom/PxMatrix #263: Scroll text within bounding box

wvthoog, 2021-03-08:
Don't know if this is possible but i want to use the scroll text function to be bound within certain dimensions. What is happening now is that on my 63x32 matrix the text scrolls from right to left over the entire display and what i want to achieve is that it's cut off at pixel 23 (calculated from the left side)

This is the current scroll text function

`void scroll_horizontal(uint8_t ypos, unsigned long scroll_delay, String text, uint8_t colorR, uint8_t colorG, uint8_t colorB)
{
    uint16_t text_length = text.length();
    display.setTextWrap(false);  // we don't wrap text so it scrolls nicely
    display.setTextSize(2);
    display.setRotation(0);
    display.setTextColor(display.color565(colorR,colorG,colorB));

    // Asuming 5 pixel average character width
    for (int xpos=matrix_width; xpos>-(matrix_width+text_length*10); xpos--)
    {
      display.setTextColor(display.color565(colorR,colorG,colorB));
      display.clearDisplay();
      display.setCursor(xpos,ypos);
      Serial.println(xpos);
      display.println(text);
      delay(scroll_delay);
      yield();

      delay(scroll_delay/5);
      yield();

    }
}`

Anyone got any idea if this is possible. If so, on how to achieve this. 

Many thanks.

davemaster, 2021-03-09:
Had You tried this code? Go ahead...

wvthoog, 2021-03-09:
@davemaster which code are you referring to ?

AlexR2106, 2021-04-07:
Hey,
Have you found a solution to this issue? I am facing the same problem as I want to display an image together with an scroll text.
Thanks

wvthoog, 2021-04-07:
Unfortunately not. Trying to achieve the same thing as you are (image + scrolling text)

Maybe another library can do the trick but haven't looked into that yet.

AlexR2106, 2021-04-07:
Nothing suitable found by now also on my side.
I started working on some code, to get the output done in a specific area. Sadly this leads being unable to use the default print command and letters need to be drawn pixel by pixel to the matrix.
I will keep you updated as soon as I have something usable.

2dom, 2021-04-07:
I think you have two options - either create a copy of the display.print function or print to some buffer and only copying the window to the actual frame buffer.

For the sencond option you could create an inline drawPixel function that allows you to write to the buffer using buffer_draw.print("Hello",color); 

```
class buffer_draw : public Adafruit_GFX {
public:
  uint8_t frame_buffer[frame_size]={0};
  buffer_draw() : Adafruit_GFX(64,64) {}

  // We need that one so we can call ".print"
  void drawPixel(int16_t x, int16_t y, uint16_t color)
  {
    
  }
};

buffer_draw buffer_drawer;
 
```

2dom, 2021-04-07:
As a reference check out https://github.com/2dom/PixelTimes/blob/master/PixelTimes.ino where I use it to create a buffer before the library had double buffer support

wvthoog, 2021-05-01:
Finally got around to test it out. And it worked !  Thank you very much

wvthoog, 2021-07-25:
Yeah I've tried to set this up myself but unfortunately my C++ skills are inferior. 

Would love to see a function in the PxMatrix library that supports Scroll Text to a particular window. 

Something like:

```
void scroll_text(uint8_t ypos, ,utin8_t maxypos, unsigned long scroll_delay, String text, uint8_t colorR, uint8_t colorG, uint8_t colorB)
```

where maxypos is the position the scroll text cuts of to the left

KvarTechnologies, 2023-02-20:
has anyone gotit?