2dom/PxMatrix #266: Old RGB matrix scan mode

outsoder, 2021-04-19:
Hello!

There is a problem connecting the old DIP RGB matrix. I can not find the right scan pattern.
The only close enough scan mode is ZIGZAG, but the first upper 4 bit block is displayed flipped.

[image]

[image]

When using display.displayTestPixel(80); the order is like this (sorry, i could not upload any video)

[image]

Maybe there is a solution with editing library?

**The code i used:**

#include <PxMatrix.h>
#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_OE 2

// Pins for LED MATRIX
PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);

// ISR for display refresh
void display_updater()
{
  //display.displayTestPattern(80);
  //display.displayTestPixel(80);
  display.display(40);
}

uint16_t myBLUE = display.color565(0, 255, 255);
uint16_t myWH = display.color565(255, 255, 255);
void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
  display.begin(4);
  display.setScanPattern(ZIGZAG);
  display.flushDisplay();
  display.setTextColor(myWH);
  display.setCursor(1,1);
  display.print("12345");
  display.setTextColor(myBLUE);
  display.setCursor(1,9);
  display.print("54321");
  Serial.println("hello");

    display_ticker.attach(0.004, display_updater);

  delay(100);
}
void loop() {
delay(10);
}

kingkumar32, 2021-05-07:
i am also having the same problem if you find the solution please share here

buxtronix, 2023-01-06:
I had weird scanning like this on a vaery similar display, and solved with `display.begin(4)` and `display.setScanPattern(ZAGGIZ)`.

Sriramsachin-6369, 2025-03-24:
I also faced same issue with p10 board, I used zagzig pattern it compatible for 32x16 board if we increase the pannel in chaining(64x32) it not supporting and split the chartered into half and print in different pannel boards