2dom/PxMatrix #275: Can I change the RGB pin?

qlalfdu, 2021-06-11:
I use ESP32 WROOM32
I want to change the RGB pins, is it possible?
Thank you.

lg007, 2021-06-12:
Hello,

What do you mean by 

> I want to change the RGB pins, is it possible?

The following snippet:

```
// Pins for LED MATRIX
#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 16
...
...

#endif
```

is where you define which ESP32 pins are used for the communication between the ESP32 and the RGB matrix. You can use any pin as long as it is an output pin on the ESP32 and it is not used by anything else. Eg. I am using pin 26 of the ESP32 for latching, therefore in my code you could see:

`#define P_LAT 26 `

There is also another pin which is used but not defined here: pin 13, which is the output of the SPI register. I don't know if this can be changed or not, they say that on the ESP32 you can almost change any pin, but I would leave that this way (it is defined in the library itself - in the PxMatrix.h file).

qlalfdu, 2021-06-13:
[image]
I want to change the RGB pins in the red square, but there is no content in the sketch.
Is there any way?

lg007, 2021-06-13:
The whole library is based on the chaining of the red, green and blue channels of the RGB display. The output of the SPI register of the ESP 32 is connected to the R1 input of the RGB display. Therefore you "feed" the display on the R1 (red 1) input channel. The output of the R1 channel is connected to the input of the R2 channel. The output of the R2 channel is connected to the input of the G1 (green 1) channel, and so on... So you will have a long shift register where channels R1, R2, G1, G2, B1 and B2 are chained together. 

The library allows you to change this order programatically with the following instruction:
`display.setColorOrder({{colororder}});`
where 
`{{colororder}} `
can be the following:
`RRGGBB, RRBBGG, GGRRBB, GGBBRR, BBRRGG, BBGGRR`

This function is needed because the quality assurance in the manufacturing of these displays is sometimes "missing"

I hope I answered your question.

qlalfdu, 2021-06-13:
I understand.
Thank you kindly explain.
Thank you lg007.

closed 2021-06-13 by qlalfdu