2dom/PxMatrix #293: None of the display.SetScanPattern() options give a proper image

zounder1, 2022-01-07:
PXMatrix version: 1.8.2
NodeMCU 1.0 ESP8266-12E 

Hi.  I am very new to LED Panels but I am making progress slowly.  I have an issue with displaying a picture on my panel - the image is mixed up slightly. 

I followed the instructions here and at https://www.instructables.com/RGB-LED-Matrix-With-an-ESP8266/

I have tried all the display.SetScanPattern(x) options with no luck. WZAGZIG2 seems to provide the closet to a correct image.  So I am hoping for advice on what my next steps should be to fix this?  Perhaps I need a new display display.SetScanPattern(x) option for my display?  That seems unlikely but I am out of ideas here.  Or maybe there is some other setting I am missing?

Thank you in advance for any advice here.  If you need any additional information, please ask and I will provide.

---Charles

Details:
My display is 64x32 1/8 scan
Product link:  https://www.aliexpress.com/item/1005002062811795.html

I downloaded a Mario image from https://github.com/witnessmenow/LED-Matrix-Display-Examples) as a way to test different scan patterns.

I only changed:
PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C);
display.begin(8);
display.setScanPattern(WZAGZIG2);

Here is the picture I get:

[image]

Here is my code for LED-Matrix-Mario-Display.  
Downloaded from:

/*******************************************************************
    Using a 64 * 32 RGB Matrix to display a mario image
 *                                                                 *
    Written by Brian Lough
    https://www.youtube.com/channel/UCezJOfu7OtqGzd5xrP3q6WA
 *******************************************************************/

// ----------------------------
// Standard Libraries - Already Installed if you have ESP8266 set up
// ----------------------------

#include <Ticker.h>

// ----------------------------
// Additional Libraries - each one of these will need to be installed.
// ----------------------------

#include <PxMatrix.h>
// The library for controlling the LED Matrix
// Needs to be manually downloaded and installed
// https://github.com/2dom/PxMatrix

Ticker display_ticker;

// Pins for LED MATRIX
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_OE 2
#define P_D 12
#define P_E 0

PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C);
// PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
// PxMATRIX display(64, 32, P_LAT, P_OE, P_A, P_B, P_C, P_D, P_E);

// Some standard colors
uint16_t myRED = display.color565(255, 0, 0);
[... 172 lines cut ...]

  display_ticker.attach(0.002, display_updater);
  yield();
  display.clearDisplay();
  drawImage(0, 0);
  delay(500);
}

void drawImage(int x, int y)
{
  int imageHeight = 32;
  int imageWidth = 64;
  int counter = 0;
  for (int yy = 0; yy < imageHeight; yy++)
  {
    for (int xx = 0; xx < imageWidth; xx++)
    {
      display.drawPixel(xx + x , yy + y, mario[counter]);
      counter++;
    }
  }
}

void loop() {
}

zounder1, 2022-01-07:
Here is output from Pattern-Test.

I only changed 
PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C);
display.begin(8);

There is no entry in this code to set:  display.SetScanPattern()
I put 
display.setScanPattern(WZAGZIG2); 
just below display.begin(8); and the screen drew thing exactly the same.  Not sure if that is expected or not!

https://user-images.githubusercontent.com/26885388/148617008-e169208e-0be7-4a8d-8a1d-4aba2ce7da0a.mov

qairulizwan, 2022-02-22:
Hi @zounder1 i think i have same problem with u.
When i test using the scan pattern, i will have same result as your video.
But if i run with the pixeltime_AVR example, i can have the good result when i set:
 display.setPanelsWidth(2);

zounder1, 2022-03-24:
Ok.  Time to circle back on this as I finally was able to get things working.  I received an email from a Github user that suggested adding the following to the 
 display.setPanelsWidth(2);

And with that option the picture of Mario shows up correctly!  

Using the sample to display Mario here are my settings:

PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C);
display.begin(8);
 **display.setPanelsWidth(2);**
display.setScanPattern(WZAGZIG2);

An important note:
I needed to download the PxMatrix from Github to get version 1.8.2 and install this manually into Arduino IDE.  The Arduino IDE manage libraries app only offers version 1.8.1 of PxMatrix for some reason.  The scan pattern WZAGZIG2 is only available in 1.8.2

zounder1, 2022-03-24:
I would like to thank qairulizwan for helping me with this!  Your message was much appreciated.

closed 2022-03-24 by zounder1