2dom/PxMatrix #303: P3 196*196 with driver ICN2037BP not worked

hasanedrees25, 2022-09-22:
I used Arduino Nano 168 and P3 from china
1. my Code was the example in Pixeltime_AVR
2.  that is the image of HUB75 , it has just ABCE , no D pin 
[image]
[image]
[image]

3- i connect it 
<html>
<body>

PI | PO
-- | --
R2 | R1
G1 | R2
G2 | G1
B1 | G2
B2 | B1

<html>
<body>

PI | NANO
-- | --
A|2
B|3
C|4
D| no D pin
E| 6
STB/LAT| 7
P_OE| 8
CLK |13
R1 |11

</body>
</html>

4.  i get nothing on the panal , when i disconnect the power from the NANO the panel show 2 lines in number 0 and number 33

fragmation, 2023-02-27:
the third pint from down (right side) is D erroneous marked on silk GND

closed 2024-03-04 by 2dom

652626737, 2024-07-21:
Hello, have you successfully driven the LED?

fragmation, 2024-07-21:
> Hello, have you successfully driven the LED?

yup the panel are working with simple gifs :D :D