2dom/PxMatrix #309: P2.5 Differences beaware

trevorboultwood, 2022-12-28:
Evening,

I have been developing a project using the PXMatrix codebase for some time and I thought it might be worth spreading some awareness on the differences between two panels I have encountered as I couldn't see much on this rep #o on this. 

The following P2.5 display works fine. Excellent! 

[https://www.aliexpress.com/item/32733178058.html?spm=a2g0o.order_list.order_list_main.11.1c301802MzxfK1](url)

Now, I bought the following screen more recently which was around two pounds cheaper (Including delivery).

[https://www.aliexpress.com/item/1005003999739439.html?spm=a2g0o.order_list.order_list_main.5.1c301802MzxfK1](url)

Of course, once again being the cheapskate I am this resulted in the super annoying problem of having to use, display.setMuxDelay(2 2, 2, 2, 2);

Wait, what 2,2,2,2,2?! Yea, this cheaper variant had to have a huge delay with each 'poll'?. I can only assume this is because it uses a far less superior driver chip. This resulted in an update rate (changing all pixels) to about a frame every 2 or or seconds. Terrible!

Frustratingly the same manufacture (sryled) sells these two displays. With the only differences within the listings being a different  driver chip, the working being JXI5020 and the not so great panel being 'Regular IC'.

Let me know if anyone has encountered a similar problem and hopefully I will sabe one of ya'll this frustrating problem.

jozsefcsiza, 2022-12-30:
Did you define the double buffer before the include of pxmatrix?
#define PxMATRIX_DOUBLE_BUFFER
#include <PxMatrix.h>

Did you set the fast update after the display begin?
display.begin(32); //Matrix Step -> 1/4, 1/8, 1/16, 1/32.
display.setFastUpdate(true);

In the loop if you use the drawpixel check the color if not black and draw it, saves time.
display.clearDisplay();
//Example if you are using a matrix to store colors.
uint16_t black = display.color565(0, 0, 0);
uint16_t color;
for (int x = 0; x < M_WIDTH; x++) {
	for (int y = 0; y < M_HEIGHT; y++) {
		color = drawMatrix[x][y];
		if (color != black)
			display.drawPixel(x, y, color);
	}
}

display.showBuffer(); //Use this after the draw if you defined the PxMATRIX_DOUBLE_BUFFER

I have a 128x64 P2 panel and I have to use display.setMuxDelay(1 1, 1, 1, 1); and it works well. I use it with an ESP32-WROOM.

trevorboultwood, 2022-12-30:
I just gave this a test and this seems to have improved the framerate considerably! Thank you very much, I actually bought another display thinking it was because of the two different types of driver chips. 

So looking at the code, i cant really seem to see why this would make that much difference? Why was the framerate so terrible without having the fast update on?

Thank you,

jozsefcsiza, 2022-12-31:
The ones who made the PxMatrix can explain the fast update purpose :)
The basic framerate optimizations are the double buffer and fast update on. After this you have to optimize your code depending on its complexity to avoid unnecessary CPU processes.

2dom, 2022-12-31:
The normal update follows the basic pattern: transfer data, enable LEDs, disable LEDs, transfer data ... and so on. Fast update leaves the LEDs enabled during transfer time greatly increasing the LEDs on-time  leading to higher brightness.

closed 2024-03-04 by 2dom