2dom/PxMatrix #310: Px matrix library for esp8266 wemos module

dileeppulluru, 2023-01-27:
Hello all,
Microcontroller - Esp8266MOD
 Am planning for P10 rgb led matrix display of bigger size 10feet width and 2 feet height which uses 30 P10 led modules of 32X16 .
 Can i use the Px matrix library for above requirement . If not which is the best library for to use.

Present i have used only two modules .
please anyone help me in understanding the  
1)display.begin(8) what is rown scan pattern mean? 
2) display.setMuxpattern
3) display.setscanpettern
          
Am not getting the perfect understanding of above lines please help me guys regarding this.
Requirement:
    1) displaying message sent through MQTT.

Thankyou in advance

2dom, 2023-01-27:
For larger displays please look into solutions that use 1: parallel data lines to drive displays, 2: more memory and processing power. I would go for Raspberry Pi - https://github.com/hzeller/rpi-rgb-led-matrix

dileeppulluru, 2023-01-30:
is it possible with ESP8266? because am restricted to use raspberry pico.If any solution please let me know

dileeppulluru, 2023-01-30:
My application has no animations only standby text and there is no need of scrolling also

closed 2024-03-04 by 2dom