2dom/PxMatrix #311: Is there a way to rotate the output 180⁰ ?

ollyf-labs, 2023-02-03:


trevorboultwood, 2023-02-05:
This could be done programmatically. Horrible algorithm and uses a lot of resources.  Effecetly, you want to rotate the 2d array of leds.

https://stackoverflow.com/questions/42519/how-do-you-rotate-a-two-dimensional-array

 void rotate180()
    {
//copy
      CRGB tmp_led[MATRIX_HEIGHT*MATRIX_WIDTH];
      for (int x = 0; x < 64; x++)
        for (int y = 0; y < 64; y++)
        {
          CRGB tmp_led[x*y] = leds[XY(x, y)];
        }
      //store
      for (int x = 0; x < 64; x++)
        for (int y = 0; y < 64; y++)
        {
          leds[XY(x, y)] = tmp_led[XY(MATRIX_HEIGHT-x, y)];
        }

2dom, 2023-02-05:
Up to now I did not see the point to rate 180deg as you can just physically rotate the panel

ollyf-labs, 2023-02-07:
> Up to now I did not see the point to rate 180deg as you can just physically rotate the panel

that is because i designed my stand to hold the panel upside down, with the hub75 connectors and the power being at the bottom half. This allowed me to hide the wires using my stand without it covering the whole panel, using much more filament for printing

closed 2024-03-04 by 2dom