2dom/PxMatrix #319: New Panel 128x64 1/32 ABC type panel with 5958 chip

Phil1972, 2023-04-01:
Hi,

It seems I ordered that panel without checking the entire details of it. So I contacted the vendor and he mentioned that this panel is driven by 5958 chips.  There is only one input on this one and so I cannot connect the output to the input to shift RGB from the output to the input as mentionned in the instructions here.

So my question is, can I make this one work?  How would I connect the RGB pins?

D,E are grounded here, so only ABC are available.

thanks for your time,

mzashh, 2023-04-19:
i would suggest trying this library instead as its much more feature rich and actively maintained https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA

mzashh, 2023-04-19:
also i would first check whether the N pins are grounded or not, if not then the one next to C is D pin and the other one is E

closed 2024-03-04 by 2dom

Phil1972, 2024-03-04:
Hi guys,

@2dom are you saying this panel is now supported? :)

thanks,