2dom/PxMatrix #325: Two displays in a chain problem

ITstreet1, 2023-06-23:
Hi,

I use this lib to drive one Waveshare 64x64 display with an ESP12F. Zero problems.
https://www.waveshare.com/wiki/RGB-Matrix-P2.5-64x64

However, when I tried to connect two in a chain, I didn't get the expected result.
I used this modified example from the lib.

```
// This is how many color levels the display shows - the more the slower the update
//#define PxMATRIX_COLOR_DEPTH 8

// Defines the buffer height / the maximum height of the matrix
//#define PxMATRIX_MAX_HEIGHT 64

// Defines the buffer width / the maximum width of the matrix
//#define PxMATRIX_MAX_WIDTH 64

// Defines how long we display things by default
//#define PxMATRIX_DEFAULT_SHOWTIME 30

// Defines the speed of the SPI bus (reducing this may help if you experience noisy images)
//#define PxMATRIX_SPI_FREQUENCY 20000000

// Creates a second buffer for backround drawing (doubles the required RAM)
//#define PxMATRIX_double_buffer true

#include <PxMatrix.h>

// Pins for LED MATRIX
#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#endif

#define matrix_width 128
#define matrix_height 64

// This defines the 'on' time of the display is us. The larger this number,
// the brighter the display. If too large the ESP will crash
uint8_t display_draw_time=50; //30-70 is usually fine

//PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);
//PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
PxMATRIX display(128,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

// Some standard colors
[... 180 lines cut ...]

  for (int xx=0; xx<16;xx++)
  {
    display.drawLine(xx+16,0,xx+16,5,display.color565(xx*16,0,0));
    display.drawLine(xx+16,6,xx+16,10,display.color565(0,xx*16,0));
    display.drawLine(xx+16,11,xx+16,15,display.color565(0,0,xx*16));
  }
  delay(1000);
  for (uint8_t dimm=255; dimm>0; dimm--)
  {
    display.setBrightness(dimm);
    delay(5);
  }
  for (uint8_t dimm=0; dimm<255; dimm++)
  {
    display.setBrightness(dimm);
    delay(5);
  }

}
```

Do I have to set something else in addition?

While driving the first one, I connected as you suggested here, and it is ok. With the second one, I connected the PO of the first to the PI of the second with a 16p IDC cable. The PO of the second I connected to the ESP12F. The connection should be ok.

ITstreet1, 2023-06-23:
The other words, not to mess up too much with this lib and displays in a chain?
With one display, the lib is great.

2dom, 2023-06-23:
Exactly...it is really meant for low pin count and not much more than 64x64 pixels

ITstreet1, 2023-06-23:
Thank you.

The one you gave a link here should do the job. Although the wiring is different.

closed 2024-03-04 by 2dom