2dom/PxMatrix #336: 32 x 16 P10 Missing row!

mailamauro888, 2024-03-13:
## Rules
1. Post your code 
**NOTE: PI connector pin n. 4, D and E are Grounded, otherwise I have a huge flickering.**
 `#include <PxMatrix.h>

#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 16
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#endif
// Pins for LED MATRIX

 PxMATRIX display(192,16,P_LAT, P_OE,P_A,P_B,P_C,P_D);//,P_E);
//PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
//PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

#ifdef ESP8266
// ISR for display refresh
void display_updater()
{
  display.displayTestPattern(70);
  // display.displayTestPixel(70);
  //display.display(70);
}
#endif

#ifdef ESP32
void IRAM_ATTR display_updater(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  //isplay.display(70);
  display.displayTestPattern(70);
  portEXIT_CRITICAL_ISR(&timerMux);
}
#endif

uint16_t myCYAN = display.color565(0, 255, 255);
void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
  display.begin(4);
    display.setScanPattern(ZAGZIG);

    display.setPanelsWidth(6);
  display.flushDisplay();
  display.setTextColor(myCYAN);
  display.setCursor(2,0);
  display.print("Pixel");
  Serial.println("hello");

  #ifdef ESP8266
    display_ticker.attach(0.002, display_updater);
  #endif

  #ifdef ESP32
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &display_updater, true);
    timerAlarmWrite(timer, 2000, true);
    timerAlarmEnable(timer);
  #endif

  delay(1000);
}

void loop() {

 delay(100);

}`
3. Post a picture of the problem and describe what you expect to see
I wish to get the missing lines!

https://github.com/2dom/PxMatrix/assets/76884805/88c36031-1dfa-4b6b-9c1d-9869e901c0c0

4. Run the pattern test and post the result as gif/video (need to have a bit more than one full RED/YELLOW/WHITE cycle)

5. State what version of PxMatrix you are running and what MicroController you use
tested with ESP32 + PxMatrix 1.7, 1.8, 1.8.1, 1.8.2. Same behaviour an all versions.

652626737, 2024-07-19:
the same problem

652626737, 2024-07-22:
> the same problem

Hi, friend, my problem has been solved, which may be caused by the connection line problem.