2dom/PxMatrix #341: I think Espressif changed ESP32 library, so we dosn't access GPIO port pins directly.

GokhanAyberk, 2024-06-13:
## Rules
1. Post your code 
2. Post a picture of the problem and describe what you expect to see
3. Run the pattern test and post the result as gif/video (need to have a bit more than one full RED/YELLOW/WHITE cycle)
4. State what version of PxMatrix you are running and what MicroController you use

Hi Dominic
I think my issue general
I am using latest version and ESP32 wroom-DA. 
My sketch was working very good. But I had this error after esp32 board lib update now,
""
/home/gokhan/Arduino/libraries/PxMatrix_LED_MATRIX_library/PxMatrix.h:83:29: error: 'GPIO' was not declared in this scope
   83 |   #define GPIO_REG_SET(val) GPIO.out_w1ts = val
      |                             ^~~~
/home/gokhan/Arduino/libraries/PxMatrix_LED_MATRIX_library/PxMatrix.h:408:5: note: in expansion of macro 'GPIO_REG_SET'
  408 |     GPIO_REG_SET(1 << _LATCH_PIN);
      |     ^~~~~~~~~~~~
""

I think Espressif changed ESP32 library, so we dosn't access GPIO port pins directly.

I tried to fix like;
#include "hal/gpio_ll.h" ; 
#include <PxMatrix.h> 

but it doesn't work. Any suggestion?
Thank you.

GokhanAyberk, 2024-06-13:
I solved my issue, The other change was in the timer object use
1) I added (#include "hal/gpio_ll.h" ;) line for direct access to GPIO pins
2) I chagged my timer usage.
void display_update_enable(bool is_enable) {
      timer = timerBegin(1000000);
      timerAttachInterrupt(timer, &display_updater);
      timerAlarm(timer, 4000, true,0);
      timerStart(timer);
 }

sseodate, 2024-12-06:
i change back FW ESP32 3.xx to 2.xx and all is ok.

namato18, 2025-04-13:
Is there any update on this? I tried adding #include "hal/gpio_ll.h" ; and I'm still getting the error 'GPIO' was not declared in this scope. Any help is appreciated!

m000lie, 2025-11-22:
thanks @namato18 ! downgrading to v2.0.16 works well 👍